Ysurac/openmptcprouter #2230: shadowsocks/IP Forwarding

Swatf, 2021-12-27:
Hi,
  - OpenMPTCProuter version: <v0.058.5>
  - OpenMPTCProuter VPS version: <0.1026.5.4.100> Debian 10
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: 
first, a big thank you for this system !

here is my problem :
I have a VPS with 2 additional IP. I can do port forwarding on the main IP but not on the others.
i try shadowsocks like the WIKI without success.

I want to be able to be assigned 1 of my additional IP to 1 local server
or
by the fierwall redirect a port of one of my additional ip to a local server

for the moment, port forwarding via firewall only works on the main IP 

config IP on the VPS :
`auto lo
iface lo inet loopback
    dns-nameservers 213.186.33.99

auto eth0
iface eth0 inet dhcp
    mtu 1500

auto eth0:0
iface eth0:0 inet static
address 54.XXX.XXX.242
netmask 255.255.255.255

auto eth0:1
iface eth0:1 inet static
address 91.XXX.XXX.196
netmask 255.255.255.255

auto eth0:2
iface eth0:2 inet static
address 91.XXX.XXX.206
netmask 255.255.255.255`
interfaces on the Router
[image]

thx !

Ysurac, 2021-12-29:
You want ip forwarding or only a port ?
What do you have in ```/etc/shorewall/rules``` on the VPS ?

Swatf, 2021-12-29:
hi idealy i want to do an ip forwarding
1 public IP = 1 VM local
the content of `/etc/shorewall/rules` :
`?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

Invalid(DROP)   net             all             tcp

DNS(ACCEPT)     $FW             net

Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn

Ping(ACCEPT)    $FW             net

Ping(ACCEPT)    net             $FW

ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535

ACCEPT          net             $FW             tcp     65222

DHCPfwd(ACCEPT) $FW             vpn

ACCEPT          net             $FW             tcp     65101   # OMR openmptcprouter open shadowsocks port tcp
ACCEPT          net             $FW             udp     65101   # OMR openmptcprouter open shadowsocks port udp
DNAT            net:10.255.255.1                vpn:$OMR_ADDR   tcp     80      -               # OMR openmptcprouter redire$DNAT            net:10.255.255.1                vpn:$OMR_ADDR   udp     80      -               # OMR openmptcprouter redire$DNAT            net             vpn:$OMR_ADDR   tcp     80      # OMR openmptcprouter redirect router 80 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     80      # OMR openmptcprouter redirect router 80 port udp`
for the moment I redirect via the firewall the main ip to a local web server.
the final objective is to be able to assign each of my local VMs a dedicated public IP 
thx for the help !

Ysurac, 2022-01-05:
There is no IP redirection here.
What is your firewall configuration in the router ? You added rules to redirect from public IP to a LAN IP ?

Swatf, 2022-01-09:
I have nothing to do with the fierwall config, I just add a port 80 redirection in order to be able to access a sit from the main ip on port 80.
I tried a lot of combinations in the fierwall a shadowocks. to try to redirect my additional ips but I was unsuccessful in redirecting one of my other IPs to a local VM.
are there solutions to try? or did I have a bug?

Ysurac, 2022-01-09:
You need to redirect ports, you need to configure that in firewall. And set public IP you want as "External IP address" in configuration.

Swatf, 2022-01-17:
hi !
ok so when i go where you told me
[image]
if i select "10.255.249.10 (gre4-oip2gre)" (coresponding to my public IP ending by "206" i have this message error " No related RPC reply" and when i refresh i have 404 error.

if i enter direcly my public ip "91.xxx.xxx.206" i have the same erro

closed 2022-04-23 by github-actions[bot]