Ysurac/openmptcprouter #2231: Support for mlx5 driver

benedikt-bartscher, 2021-12-27:
Hello there, i need a driver for my Mellanox ConnectX-4 network card. I am passing it via SRV-IOV to my openmptcprouter VM. I did some research and found the OpenWRT package "kmod-mlx5-core". Could you please add this package to the Openmptcprouter Repo so everyone can install it easy?

Keep the good work up, thank you very much!

Ysurac, 2021-12-29:
The package kmod-mlx5-core is available in System->Software after updating list.

benedikt-bartscher, 2021-12-29:
Probably not for bleeding edge versions, right? I get this error message:
"The installed version of package kernel is not compatible, require 5.4.106-1-53b4fb15… while 5.4.132-1-4282eeab… is installed."

benedikt-bartscher, 2021-12-30:
I just tested the official release and the driver installed fine.
Sadly i am stuck on a boot loop now. While loading mlx5 on boot the QEMU VM directly crashes and reboots. I have SRV-IOV setup and all other VMs are doing fine with these settings. Any advice?

benedikt-bartscher, 2021-12-30:
[image]

benedikt-bartscher, 2021-12-30:
I just tried another Card from HP (qlogic chip). It needs kmod-qlcnic which is sadly not included in openmptcprouter :(

closed 2022-04-05 by github-actions[bot]