Ysurac/openmptcprouter #2234: Can't forward port

albertoZurini, 2022-01-04:
## Expected Behavior
VPS port 12345 is forwarded to host 192.168.1.33 port 443. Basically this: `ssh -R \*:12345:0.0.0.0:443`.

## Current Behavior
The forward doesn't work with both VPN and v2ray configurations and I can't figure out why. I've followed the guide, but the debug section didn't help much:
*  The content of the files is fine
    * `/etc/shorewall/rules`: `DNAT            net             vpn:$OMR_ADDR   tcp     12345   # OMR openmptcprouter redirect router 12345 port tcp`
    * `/etc/shorewall/params.vpn`: ```VPS_ADDR=10.255.255.1
VPS_IFACE=gt-tun0
OMR_ADDR=10.255.255.2```
* TCPdump log is empty on `tun0`, `tun1` and `gt-tun0` and the port too

### V2ray conf:
[image]

### VPN conf:
[image]

[image]

I've also tried swapping the ports but still can't get any log.

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: has no restrictions
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-01-04:
Try tcpdump on the VPS, with something like ```tcpdump -i any -p port 12345``` and check that you get something.

albertoZurini, 2022-01-04:
I think I've managed to solve, I just had to set this only rule and nothing else:
[image]

The problem now is that I can't access this port forward from a docker container (into the compose) that is being run on the VPS. Also I'm facing issues when launching `docker-compose up` because sometimes `iptables` doesn't work correctly, is there any way to block openmptcp from accessing ports 80 and 443?

Thanks a lot for your support.

closed 2022-04-11 by github-actions[bot]