Ysurac/openmptcprouter #2235: Best configuration for low latency and seamless failover

jamesdiener, 2022-01-05:
## Expected Behavior

I would like to know what configurations I have to make in order to optimize for low latency and seemless failover

## Current Behavior
The latency isn't better than without Open MPTCP Router

## Specifications

  - OpenMPTCProuter version: 
  - openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2711-rpi-4-ext4-factory.img
  - OpenMPTCProuter VPS version: 
  - 0.1026
  - OpenMPTCProuter VPS provider:
  - netcup
  - OpenMPTCProuter platform: 
  - RPI4

suyuan168, 2022-01-05:
Please check the wiki~

jamesdiener, 2022-01-05:
@suyuan168 I checked and found nothing about optimization for best latency

jamesdiener, 2022-01-05:
But if you found something please share the link :)

mstrolo, 2022-01-05:
also wants to know!

mstrolo, 2022-01-05:
if I use v2ray instead of Shadowsocks it is now using the glorytun?

for me Shadowsocks is not working it saying cant get IP address so i am using v2ray on intel nuc

jamesdiener, 2022-01-05:
> There is no best, only better. Expand the page according to the wiki。
> https://github.com/Ysurac/openmptcprouter/wiki/Optimization

As far as I know an optimization of latency requires a different configuration than optimization of bandwith/aggregation. That's why I would like to know which parameters I have to change or check. And I assume Congestion Control and tcp syn retries as described in the wiki are not the only relevant parameters. Further there is no information which congestion control promises better latency.

⁣Get BlueMail for Android ​

Ysurac, 2022-01-05:
You can't get a better latency than your best connection.
For congestion control, you have to test: each case is different.

jamesdiener, 2022-01-06:
That's what I tried to explain in my original issue. The latency is often worse than the latency of the best connection. I assume that the router doesn't utilize that connection, but also the connection with higher latency. That's why I want to know: how to optimize the system for best latency instead of best throughout/bandwith?

⁣Get BlueMail for Android ​

jamesdiener, 2022-01-06:
Apparently ReMDeLa protocol can achieve good latency:

https://www.researchgate.net/publication/336434800_Redundant_Multipath-TCP_Scheduling_with_Desired_Packet_Latency

⁣Get BlueMail for Android ​

mstrolo, 2022-01-06:
there are two that suppose to make it happed
 SlothTail and ReMDeLa
how do we use them?

jamesdiener, 2022-01-06:
CFS is another candidate. Have to check again if one of them is already available in OMR.

⁣Get BlueMail for Android ​

Adorfer, 2022-01-06:
so what is the wiki-article we are talking here? what is the current recommendation for e.g. 3 similar, stable WAN connections via broadband.

jamesdiener, 2022-01-07:
Suyuan referred to this page:

https://github.com/Ysurac/openmptcprouter/wiki/Optimization

But it doesn't say much more than 'you can try another congestion control algorithm', and it doesn't say what the effect of such optimization would be.

Quote of the article:

"To optimize TCP traffic

In Network->MPTCP, you can try another TCP Congestion Control. BBR2 seems to offer a good result in many case (available only on 64bits arch), BBR give a good result only when connection are good without packets lost. Cubic work in every case but can be slower.

If you have a bad connection, you can increase TCP Syn retries."

⁣Get BlueMail for Android ​

mstrolo, 2022-01-07:
how do we add those protocols to omr?

jamesdiener, 2022-01-10:
ECF is available in MPTCP and seems to do a good job in terms of lateny

⁣Get BlueMail for Android ​

jamesdiener, 2022-01-10:
ECF is available in MPTCP and seems to do a good job in terms of latency

⁣Get BlueMail for Android ​

mstrolo, 2022-01-10:
what does ecf opyimize?
latency?
speed?

jamesdiener, 2022-01-10:
It means, ECF (Earliest Completion First)

Better latency is one benefit compared to standard mptcp

⁣Get BlueMail for Android ​

jamesdiener, 2022-01-10:
Good discussion:

https://github.com/Ysurac/openmptcprouter/issues/972

⁣Get BlueMail for Android ​

mstrolo, 2022-01-15:
did you manage to understand what will be best for each conifguration?
like what's best for latency
or what it best for ping or streaming?

jamesdiener, 2022-01-21:
ECF improves latency a lot in a heterogen wan environment as it will prioritize the low latency connection. So it improved my connection a lot for rdp application which requires low latency. For uninterrupted live streaming, I'm still trying to figure out which congestion control works best in my scenario. I'd need that for solid video calls

mstrolo, 2022-01-21:
what I am looking to achieve is 3 options

one for speed optimization
one for latency 
and one for packets which is the video and that

jamesdiener, 2022-01-21:
Interesting targets! And is there a way to switch between those 3 'profiles' in a convenient way?

mstrolo, 2022-01-21:
i did it with a phone qpp

jamesdiener, 2022-01-21:
What is qpp?

jamesdiener, 2022-01-21:
Phone app? Which app did you use?

mstrolo, 2022-01-31:
Blynk
And i added options to change the Congestion.

Do you have an idea for duplicate packets? I know it can bring down the delay on zoom and more like that

mstrolo, 2022-03-27:
hi how can we deal with video buffering?
 i have founs this:
Cross-Layer Scheduler for Video Streaming over MPTCP

is this Scheduler part of the current version?

Ysurac, 2022-03-27:
No, but if source code is available I may be able to add it.

mstrolo, 2022-03-27:
https://kipdf.com/cross-layer-scheduler-for-video-streaming-over-mptcp_5aed39187f8b9a932c8b460a.html
this is what i found

TalalMash, 2022-03-28:
You can use the integrated ndpi and shorewall to route traffic <4Mbit and video streams to Engarde. 
Dynamic master WAN selection is still needed for ~3 seconds failover for bulk downloads so it's not seamless for large bandwidth, not a problem usually as it's a non real time connection. This assumes that the secondaries are lossy too as losing the master is usually not an issue and seamless.

mstrolo, 2022-03-30:
didnt unddrstand a thing hhh

TalalMash, 2022-05-21:
Just tested latest, seamless failover is working with openmptcprouter v0.59beta6-5.4 r0+16591-b4ea8e1089
A milestone passed :)

closed 2022-08-24 by github-actions[bot]