Ysurac/openmptcprouter #2236: Aggregation stopped working automatically

k4mrul, 2022-01-08:
## Expected Behavior
Aggregation stopped working after a few minutes. It needs to be up all the time

## Current Behavior
Currently the bonding/aggregation works fine for a few minutes. But after some time (1 or 2 minutes), the aggregation stops working and only master WAN is used. The only way to fix this issue to reboot the router/rpi4. After reboot, aggregation works fine again for few minutes, then again it stopped working. Only master wan used.

This issue is related with https://github.com/Ysurac/openmptcprouter/issues/958. I tried the solutions mentioned there but still, issue persist. Also tried other VPNs (mlvpn, glorytun, dsvpn) all works fine but stopped working again after a few mins.

Here are the system logs (Multipath Debug enabled)
https://paste.ee/p/zJXWs

Another system log (Multipath Debug disabled)
https://paste.ee/p/juACn

MPTCP Fullmesh
```
Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 6
3, 4, 0, 192.168.0.150, 16
4, 5, 0, 192.168.1.164, 17
IPv6, next v6-index: 1
```

[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: Azure
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.4

This is an amazing project btw! Will definitely support this project when it will work without issue

Ysurac, 2022-01-09:
You should keep glorytun-tcp for now, and set SQM with download and upload speed in wizard.

k4mrul, 2022-02-22:
I set SQM and keep glorytun-tcp. Issue didn't solve :( only master connection works after few minutes of reboot.

closed 2022-05-29 by github-actions[bot]