Ysurac/openmptcprouter #2238: two connections with the same internet address blocks OMR

sonnyxp2008, 2022-01-08:
## Expected Behavior
 Hi, I tried to join 3 connections of which two are of the same wan address because I take it from two different point points. Basically I wanted to take all the available bandwidth through two antennas. The final result is the total blocking of the program and therefore of the connection. Can anyone help me? 
use as ip address 192.168.1.147
192.168.2.150
192.168.6.150
192.168.3.140
the second address with the fourth are on the same line and therefore have the same internet address

## Current Behavior
connection block and OMR total block

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: ARUBA CLOUD
  - OpenMPTCProuter platform: x86_64

sonnyxp2008, 2022-01-09:
nessuno riesce ad aiutarmi?

Ysurac, 2022-01-09:
I need more info to find what is the problem exactly. Total blocking is due to a network configuration issue, so I need more network configuration info.

closed 2022-04-14 by github-actions[bot]