Ysurac/openmptcprouter #2239: server recv: Operation timed out

sonnyxp2008, 2022-01-09:
## Expected Behavior
several errors of this type server recv: Operation timed out
Sun Jan  9 12:00:42 2022 daemon.err /usr/bin/ss-redir[31580]: send: Connection reset by peer
Sun Jan  9 12:01:02 2022 daemon.err /usr/bin/ss-redir[31578]: server recv: Operation timed out
Sun Jan  9 12:01:03 2022 daemon.err /usr/bin/ss-redir[31577]: send: Connection reset by peer
Sun Jan  9 12:01:24 2022 daemon.err /usr/bin/ss-redir[31578]: server recv: Operation timed out
Sun Jan  9 12:03:29 2022 daemon.err /usr/bin/ss-redir[31580]: send: Connection reset by peer
Sun Jan  9 12:04:21 2022 daemon.err /usr/bin/ss-redir[31579]: send: Connection reset by peer
Sun Jan  9 12:04:27 2022 daemon.err /usr/bin/ss-redir[31579]: send: Connection reset by peer
Sun Jan  9 12:04:27 2022 daemon.err /usr/bin/ss-redir[31579]: send: Connection reset by peer
Sun Jan  9 12:04:29 2022 daemon.err /usr/bin/ss-redir[31578]: server recv: Operation timed out
Sun Jan  9 12:04:50 2022 daemon.err /usr/bin/ss-redir[31577]: send: Connection reset by peer
Sun Jan  9 12:05:17 2022 daemon.err /usr/bin/ss-redir[31577]: send: Connection reset by peer
Sun Jan  9 12:06:29 2022 daemon.err /usr/bin/ss-redir[31577]: send: Connection reset by peer
Sun Jan  9 12:07:12 2022 daemon.err /usr/bin/ss-redir[31577]: send: Connection reset by peer
Sun Jan  9 12:07:18 2022 daemon.err /usr/bin/ss-redir[31578]: send: Connection reset by peer
Sun Jan  9 12:07:48 2022 daemon.err /usr/bin/ss-redir[31580]: send: Connection reset by peer
Sun Jan  9 12:07:49 2022 daemon.err /usr/bin/ss-redir[31580]: server recv: Operation timed out
Sun Jan  9 12:08:02 2022 daemon.err /usr/bin/ss-redir[31577]: send: Connection reset by peer
Sun Jan  9 12:08:52 2022 daemon.err /usr/bin/ss-redir[31580]: send: Connection reset by peer
Sun Jan  9 12:09:32 2022 daemon.err /usr/bin/ss-redir[31577]: send: Connection reset by peer
Sun Jan  9 12:10:02 2022 daemon.err /usr/bin/ss-redir[31580]: send: Connection reset by peer
Sun Jan  9 12:10:28 2022 daemon.err /usr/bin/ss-redir[31579]: send: Connection reset by peer
Sun Jan  9 12:10:39 2022 daemon.err /usr/bin/ss-redir[31578]: send: Connection reset by peer
Sun Jan  9 12:11:06 2022 daemon.err /usr/bin/ss-redir[31579]: server recv: Operation timed out
Sun Jan  9 12:11:21 2022 daemon.err /usr/bin/ss-redir[31580]: send: Connection reset by peer
Sun Jan  9 12:13:05 2022 daemon.err /usr/bin/ss-redir[31578]: send: Connection reset by peer
Sun Jan  9 12:13:40 2022 daemon.err /usr/bin/ss-redir[31579]: send: Connection reset by peer
Sun Jan  9 12:14:04 2022 daemon.err /usr/bin/ss-redir[31578]: send: Connection reset by peer
Sun Jan  9 12:14:15 2022 daemon.err /usr/bin/ss-redir[31577]: send: Connection reset by peer
Sun Jan  9 12:14:46 2022 daemon.err /usr/bin/ss-redir[31580]: server recv: Host is unreachable
Sun Jan  9 12:15:19 2022 daemon.err /usr/bin/ss-redir[31580]: send: Connection reset by peer
Sun Jan  9 12:15:20 2022 daemon.err /usr/bin/ss-redir[31579]: send: Connection reset by peer
Sun Jan  9 12:15:36 2022 daemon.err /usr/bin/ss-redir[31578]: server recv: Operation timed out
Sun Jan  9 12:15:44 2022 daemon.err /usr/bin/ss-redir[31578]: send: Connection reset by peer
Sun Jan  9 12:17:01 2022 daemon.err /usr/bin/ss-redir[31579]: remote recv: Connection reset by peer
Sun Jan  9 12:17:44 2022 daemon.err /usr/bin/ss-redir[31578]: send: Connection reset by peer
Sun Jan  9 12:17:45 2022 daemon.err /usr/bin/ss-redir[31579]: server recv: Operation timed out
Sun Jan  9 12:17:47 2022 daemon.err /usr/bin/ss-redir[31577]: send: Connection reset by peer
Sun Jan  9 12:18:08 2022 daemon.err /usr/bin/ss-redir[31577]: send: Connection reset by peer
Sun Jan  9 12:18:36 2022 daemon.err /usr/bin/ss-redir[31578]: server recv: Operation timed out
Sun Jan  9 12:18:44 2022 daemon.err /usr/bin/ss-redir[31578]: remote recv: Connection reset by peer

## Current Behavior
i can't find the problem

## Specifications

  - OpenMPTCProuter version:  v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: ARUBA CLOUD
  - OpenMPTCProuter platform:  x86_64

Ysurac, 2022-01-09:
These errors mean a TCP timeout, not always a problem, only something unreachable or that close connection often.
Can be due to some not responding IP in omr-tracker (I remove some IPs that don't answer anymore at each update).

cagnulein, 2022-02-05:
@sonnyxp2008 same here and i have the same provider, ARUBA CLOUD. Are you italian too? Did you try with OVH?

sonnyxp2008, 2022-02-05:
Yes, I'm Italian. I have not tried OVH because I have not found Italian vps. I have tried other providers but have not had satisfactory results

cagnulein, 2022-02-05:
@sonnyxp2008 so you're still with aruba? or you abandoned this project?

sonnyxp2008, 2022-02-05:
yes i am using aruba

cagnulein, 2022-02-05:
do you have a stable ping? every 10-15 seconds i got a 150-200ms ping and i thought it was about these errors @sonnyxp2008

Ysurac, 2022-02-05:
Ping doesn't use shadowsocks. Ping use the VPN, you can choose another VPN if you want a better ping but in most case latency is measured using TCP. If you have an issue in TCP latency then you should enable SQM in the wizard with a download and upload speed set. You can also choose another TCP Congestion Control in Network->MPTCP.

sonnyxp2008, 2022-02-05:
I use the datacenter dc 3 bergamo in Aruba. I have no problems with latency despite being high for me. it is around 37 - 39 ms The congestion I use is Cubic. I've tried BBR and BBR2 and I'm having problems with stability and latency

sonnyxp2008, 2022-02-05:
in all honesty I have tried Amazon in the past and it is definitely better but the costs soar

closed 2022-05-12 by github-actions[bot]