Ysurac/openmptcprouter #2241: DNS requests

DigibitDK, 2022-01-12:
This is a general question:

DNS requests seems to be routed trough the MPTCP-system to the server-end be default, and that slows down the connection, as well as making some GEO-issues due to the fact, that client is in another country than the server.

So, my question is:
How can I change the client-MPTCP unit to use the local DNS-serveres provided by the connected WANs directly? (Bypass DNS-requests from using MPTCP-system)

If possible, send the request on all WANs, and use the fastest answer (Parallel-request).

## Specifications
Latest software to date.

Ysurac, 2022-01-12:
There is many problems when using ISP DNS:
* They can lie (In many country ISP DNS are used for censorship)
* They can use DNS GeoIP and route from the server can be longer
* They can use a too long cache
* they can be down
* ...

OpenMPTCProuter Server part should be in same country as the user, else latency is often higher.
If you really want to use ISP DNS, you can add them in Network->DHCP&DNS and in omr-bypass, but it's often a bad idea.

DigibitDK, 2022-01-12:
Yes, i'm aware for the facts.
But since the ISP (Server) is at OVH in Germany, and i'm in Denmark, this creates a lot of problems since Google etc. thinks im in France (Even that the server is in Germany, the IP is located in France.

The latency (distance between client and server) is not that important here, since it my setup is used for failover/reliability  more than the need for speed. Its kind of slow yes, but that is acceptable in this case.

Ill take a look at your DNS-solution :) Thanks for reply.

Note: Ill been testing your EdgeRouter X image, and yes I know its in Alpha-state, and it sure has some bugs here and there. Let me know if you need any feedback, otherwise ill wait until beta-level :)

Adorfer, 2022-01-13:
In case the exit-IP for your DNS is different from the exit-IP of your http-Traffic, then you might get totally bad (slow routing) CDN servers.

closed 2022-04-20 by github-actions[bot]