Ysurac/openmptcprouter #2242: MPTCP seems to be blocked

patimages, 2022-01-12:
[image]
[image]
Hello, 
I have an issue with MPTCP (worked well for years) and it does seem that the MPCTP is not blocked by the provider so what could be my problem? FYI, I am on the latest versions of server and client. 
Merci encore !

Ysurac, 2022-01-12:
Please use the template when you open an issue.
As you can see with MPTCP Support Check it's a false issue.

patimages, 2022-01-12:
Sorry for missing out on the template (it has been a while since I had issues, I forgot the process: you did too good of a job :-D). 
This false issue leads to a dramatically reduced bandwidth (most of the bandwidth goes trough the slow DSL interface), any advice to recover ?

patimages, 2022-01-12:
PS : the alerts just disappeared ... Any idea what was the problem ?

Ysurac, 2022-01-12:
They way the test is now made may result in this problem if some packets are lost for example.

patimages, 2022-01-12:
Got it : thanks again for the fastest support ever !

closed 2022-04-18 by github-actions[bot]