Ysurac/openmptcprouter #2243: no documentation for Banana Pi BPI-R64

abismailinglist, 2022-01-12:
I've been having some problems trying to install OpenMPTCPRouter on my Banana Pi BPI-R64. I've noticed that, even though there are some images for the Banana Pi BPI-R64 in the downloads section, there is no mention of it in the documentation. Would you mind writing a few lines about how to proceed? I'm able to get to the point where it asks me from the .ibn file but I don't know how to provide it, Google suggests that I should do that with tftp but the Banana PI's Ethernet ports apparently don't seem to work at that stage.

abismailinglist, 2022-01-13:
To be more clear, I'm stuck here, in U-boot. I can't seem to get Ethernet or USB ports to work, so I don't know how I'm supposed to provide the .itb file.
[image]

Ysurac, 2022-01-15:
Installation is the same as openwrt, so https://openwrt.org/toh/sinovoip/bananapi_bpi-r64_v1.1