Ysurac/openmptcprouter #2246: Versioning & OpenWRT Builds?

dm17, 2022-01-15:
I'm trying to find information on which OpenWRT version is included in the latest OpenMPTCRouter. Does it keep up with OpenWRT versions (current stable being v19.07). The about page says OpenWRT is "included," but I don't know in what capacity.

Is there a way to download the packages directly or must we trust the released images from OpenMPTCRouter? It is nice to be able to bake our own images for the sake of trust / good practice.

Thanks for the great project!

Ysurac, 2022-01-15:
On current release, a snapshot release is included in OpenMPTCProuter. You can find OpenWrt branch used in https://github.com/Ysurac/openmptcprouter/blob/v0.58.5/build.sh
On next release this will be 21.x openwrt release for 5.4 kernel (develop branch for 5.15 kernel).
I clone all OpenWRT packages to keep a "stable" release.

dm17, 2022-01-15:
@Ysurac Thanks! Does that mean most of the OpenWRT docs apply to OpenMTCPRouter?

closed 2022-04-22 by github-actions[bot]