Ysurac/openmptcprouter #2247: new version not working at all

mstrolo, 2022-01-16:
## Expected Behavior

Omr alpha 0.59 upgrade working

## Current Behavior

i have nothing that can connect 
cannot connect to the vps server and not to any LTE modem i have connected
[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+18442-02de391b08
  - OpenMPTCProuter VPS version: 0.1027-test 5.4.132-mptcp
  - OpenMPTCProuter VPS provider: y on server that i rented 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-01-16:
I don't know what you mean, there is no new release.
If you mean snapshot release, there is 2 snapshot based on 2 kernel that doesn't work the same way.

mstrolo, 2022-01-16:
yes I have installed the snapshot

can you explain what does that means?

2 kernels

do you have any idea why its not working for me now?

Ysurac, 2022-01-16:
no, I don't know what version you downloaded.
Snapshot are not mean to be stable, it's snapshot ;)

mstrolo, 2022-01-16:
openmptcprouter v0.58.5 r0+18442-02de391b08
that one

so if i want to go back to the server version and openmptcp version that was stable just downgrade and run the script n the vps?

mstrolo, 2022-01-16:
VPS version is 0.1027-test

can i downgrade it?

mstrolo, 2022-01-16:
this is where am at now:

[image]

Ysurac, 2022-01-16:
Yes you should use normal VPS script.
What is strange is that wan address you get are Private address...

mstrolo, 2022-01-16:
from the lte modems?

mstrolo, 2022-01-17:
now its good but i have other issue

every time am runing a test something else falls down

my log:

[image]

Ysurac, 2022-01-17:
You may need to enable SQM with download and upload speed in wizard.

mstrolo, 2022-01-17:
> You may need to enable SQM with download and upload speed in wizard.

did that but yet i have one of them falling down:

[image]

[image]

[image]

mstrolo, 2022-01-17:
changed the SQM and yet
[image]

hairy-dog, 2022-01-20:
There is something wrong with how your network settings are. On my system, I have wan1 with an IP address of 192.168.14.10, wan2 on 192.168.12.10 and want3 on 192.168.10.10 - each of them has a wan address allocated by their remote network.
In your earlier screen captures, your wan addresses were local addresses, which can never work. Now they look correct, but your ip addresses for them are in different ranges.
I suggest that you put that right first.

closed 2022-04-26 by github-actions[bot]