Ysurac/openmptcprouter #2250: Wans being switched off

slazinger7, 2022-01-18:
After setup using the Wizard, the system periodically (about every 4-6 minutes) disconnects both Wans and I see the following errors in the logs:

Tue Jan 18 01:46:08 2022 user.notice post-tracking-post-tracking: wan2 (wan2) switched off because check error and ping from 192.168.11.2 error (4.2.2.1,8.8.8.8,80.67.169.12)
Tue Jan 18 01:46:08 2022 user.notice post-tracking-post-tracking: Delete default route via 192.168.11.1 dev wan2
Tue Jan 18 01:46:11 2022 user.notice post-tracking-post-tracking: wan1 (wan1) switched off because check error and ping from 192.168.13.2 error (4.2.2.1,8.8.8.8,80.67.169.12)
Tue Jan 18 01:46:11 2022 user.notice post-tracking-post-tracking: Delete default route via 192.168.13.1 dev wan1

IPv6 and DHCP is turned off on the modems (one is a MoFi and the second is running Rooter/GoldenOrb).

I understand there may be some sort of communication error, but I'm not sure how to troubleshoot.  Any help would be greatly appreciated.

## Specifications
  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: Ubuntu
  - OpenMPTCProuter VPS provider: Self
  - OpenMPTCProuter platform: RPI4B

Ysurac, 2022-01-18:
As you can see, the ping from wan1 and wan2 to internet is not working.
You can try yourself: ```ip -B -I wan1 8.8.8.8``` and ```ip -B -I wan2 8.8.8.8```
Check that ping is not blocked by ISP or modems.

mstrolo, 2022-01-19:
i have the same issue 

i have ping and all but the wans just keep droping

slazinger7, 2022-01-21:
I think the ping is being blocked by the modem - it's a mofi 4500. Any idea what setting to change?

I have a separate modem connected to a raspberry pi 4 running ROOTer, and I don't have any disconnect issues.

slazinger7, 2022-01-21:
I tried flashing the mofi 4500 with Rooter, but no luck.  Perhaps Visible (the carrier) could be blocking the ping.  Is there a way around this?

Ysurac, 2022-01-22:
In Services->OMR-Tracker you can change the type of connection status check.

slazinger7, 2022-01-22:
Switching OMR-Tracker from "ping to "none" seems to have resolved the issue.  Je vous remercie!

cagnulein, 2022-02-07:
i'm having the same issue. I just switched to none in the Default settings in omr-tracker (leaving ping in the ORMVPN). It's the same thing that you did @slazinger7 ?

slazinger7, 2022-02-24:
These logs look different than mine; I wasn't getting any "daemon.err" notifications.

cagnulein, 2022-02-24:
your solution doesn't work for me indeed @slazinger7 :)
i'm waiting the next release from @Ysurac i guess it will solve this too

closed 2022-05-30 by github-actions[bot]

alchool, 2023-09-15:
It stops to miss checks also if u disable ss or proxy