Ysurac/openmptcprouter #2252: wan or server keep turning off

mstrolo, 2022-01-21:
## Expected Behavior

when I am connecting all the Wans and the server they appear green and all fine
[image]

## Current Behavior
every time I am running a speed test something turns down and the system turns it off.

[image]
[image]
[image]

[image]

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2022-01-24:
As you can see in log, ping from 10.100.102.90 to 1.1.1.1 for example is not working.
You can check yourself using "ping -B -I 10.100.102.90 1.1.1.1"
Check that ping is allowed, if it's not the case use another method in Services->OMR-Tracker

mstrolo, 2022-01-24:
PING 1.1.1.1 (1.1.1.1) from 10.100.102.90 : 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=58 time=15.4 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=58 time=14.8 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=58 time=14.1 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=58 time=14.5 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=58 time=14.8 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=58 time=16.0 ms
64 bytes from 1.1.1.1: icmp_seq=7 ttl=58 time=16.6 ms
64 bytes from 1.1.1.1: icmp_seq=8 ttl=58 time=14.2 ms

Ysurac, 2022-01-24:
At least sometimes packets are lost.
You can try to disable "Server http test" in Services->OMR-Tracker, this may help for wan2.
You should also update to latest stable release.

mstrolo, 2022-01-24:
so i need to add wan2 in the omr-tracker
and disable the server http test?

closed 2022-04-30 by github-actions[bot]