Ysurac/openmptcprouter #2253: VPS Server Script Fails

pbix, 2022-01-23:
Error free VPS configuration

## Current Behavior

I understand this this bleeding edge but thought that I should report that the VPS script referenced for snapshots does not appear to run on Debian 10.

Running the server install script referenced here [https://github.com/Ysurac/openmptcprouter/wiki/Snapshots](url) on a fresh install of Debian 10 results in an error which aborts the configuration.
## Possible Solution

## Steps to Reproduce the Problem

  1. Create a virgin Debian 10 VPS
  2. Run the install script "wget -O - http://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh"

## Context (Environment)

--Results shown on the command line ----
Remove lock and update packages list...
Hit:1 http://mirrors.linode.com/debian buster InRelease
Get:2 http://mirrors.linode.com/debian-security buster/updates InRelease [65.4 kB]
Get:3 http://mirrors.linode.com/debian buster-updates InRelease [51.9 kB]
Get:4 http://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Hit:5 https://repo.openmptcprouter.com next InRelease
Hit:6 https://repo.openmptcprouter.com buster InRelease
Fetched 164 kB in 1s (163 kB/s)
Reading package lists... Done
Install apt-transport-https, gnupg and openssh-server...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
apt-transport-https is already the newest version (1.8.2.3).
gnupg is already the newest version (2.2.12-1+deb10u1).
openssh-server is already the newest version (1:7.9p1-10+deb10u2).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up omr-glorytun-tcp (0.0.35-4) ...
Job for glorytun-tcp@tun0.service failed because the control process exited with error code.
See "systemctl status glorytun-tcp@tun0.service" and "journalctl -xe" for details.
dpkg: error processing package omr-glorytun-tcp (--configure):
 installed omr-glorytun-tcp package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 omr-glorytun-tcp
E: Sub-process /usr/bin/dpkg returned an error code (1)

-----Results from journalctl -xe
root@localhost:/var/log# journalctl -xe
-- A start job for unit glorytun-tcp@tun0.service has finished with a failure.
-- 
-- The job identifier is 57164 and the job result is failed.
Jan 23 16:41:29 localhost systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Jan 23 16:41:29 localhost systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 9.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Automatic restarting of the unit glorytun-tcp@tun0.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Jan 23 16:41:29 localhost systemd[1]: Stopped Glorytun TCP on tun0.
-- Subject: A stop job for unit glorytun-tcp@tun0.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A stop job for unit glorytun-tcp@tun0.service has finished.
-- 
-- The job identifier is 57222 and the job result is done.
Jan 23 16:41:29 localhost systemd[1]: Starting Glorytun TCP on tun0...
-- Subject: A start job for unit glorytun-tcp@tun0.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit glorytun-tcp@tun0.service has begun execution.
-- 
-- The job identifier is 57222.
Jan 23 16:41:29 localhost systemd[1]: glorytun-tcp@tun0.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit glorytun-tcp@tun0.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 1.

=====Examining /var/log/daemon I find many of the following records.=====
Jan 23 16:35:27 localhost systemd[1]: Starting Glorytun TCP on tun0...
Jan 23 16:35:27 localhost systemd[1]: glorytun-tcp@tun0.service: Main process exited, code=exited, status=1/FAILURE
Jan 23 16:36:57 localhost systemd[1]: glorytun-tcp@tun0.service: Start-post operation timed out. Stopping.
Jan 23 16:36:57 localhost systemd[1]: glorytun-tcp@tun0.service: Control process exited, code=killed, status=15/TERM
Jan 23 16:36:57 localhost systemd[1]: glorytun-tcp@tun0.service: Failed with result 'exit-code'.
Jan 23 16:36:57 localhost systemd[1]: Failed to start Glorytun TCP on tun0.
Jan 23 16:36:57 localhost systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Jan 23 16:36:57 localhost systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 6.
Jan 23 16:36:57 localhost systemd[1]: Stopped Glorytun TCP on tun0.
Jan 23 16:36:57 localhost systemd[1]: Starting Glorytun TCP on tun0...
Jan 23 16:36:57 localhost systemd[1]: glorytun-tcp@tun0.service: Main process exited, code=exited, status=1/FAILURE

## Specifications
I am using Linode to host this VPS.

pbix, 2022-02-05:
Some more details on what I am seeing.  This time on a Debian 11 VPS.

Setting up omr-glorytun-tcp (0.0.35-4) ...
Job for glorytun-tcp@tun0.service failed because the control process exited with error code.
See "systemctl status glorytun-tcp@tun0.service" and "journalctl -xe" for details.
dpkg: error processing package omr-glorytun-tcp (--configure):
 installed omr-glorytun-tcp package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 omr-glorytun-tcp
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@localhost:~# systemctl status glorytyn-tcp@tun0.service
Unit glorytyn-tcp@tun0.service could not be found.
root@localhost:~# systemctl status glorytun-tcp@tun0.service
● glorytun-tcp@tun0.service - Glorytun TCP on tun0
     Loaded: loaded (/lib/systemd/system/glorytun-tcp@.service; enabled; vendor preset: enabled)
     Active: activating (start-post) (Result: exit-code) since Sat 2022-02-05 15:47:01 UTC; 1min 9s ago
    Process: 19096 ExecStart=/usr/bin/glorytun-tcp-run /etc/glorytun-tcp/tun0 (code=exited, status=1/FAILURE)
   Main PID: 19096 (code=exited, status=1/FAILURE); Control PID: 19097 (post.sh)
      Tasks: 2 (limit: 1134)
     Memory: 320.0K
        CPU: 80ms
     CGroup: /system.slice/system-glorytun\x2dtcp.slice/glorytun-tcp@tun0.service
             ├─19097 /bin/sh /etc/glorytun-tcp/post.sh /etc/glorytun-tcp/tun0
             └─19215 sleep 2

Feb 05 15:47:01 localhost systemd[1]: Starting Glorytun TCP on tun0...
Feb 05 15:47:01 localhost systemd[1]: glorytun-tcp@tun0.service: Main process exited, code=exited, status=1/FAILURE
root@localhost:~# journalctl -xe
Feb 05 15:48:32 localhost sshd[19285]: pam_unix(sshd:auth): authentication failure; logname= uid=0 euid=0 tty=ssh ruser= rhost=92.255.85.237
Feb 05 15:48:32 localhost systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 8.
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ Automatic restarting of the unit glorytun-tcp@tun0.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Feb 05 15:48:32 localhost systemd[1]: Stopped Glorytun TCP on tun0.
░░ Subject: A stop job for unit glorytun-tcp@tun0.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A stop job for unit glorytun-tcp@tun0.service has finished.
░░ 
░░ The job identifier is 3808 and the job result is done.
Feb 05 15:48:32 localhost systemd[1]: glorytun-tcp@tun0.service: Start request repeated too quickly.
Feb 05 15:48:32 localhost systemd[1]: glorytun-tcp@tun0.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit glorytun-tcp@tun0.service has entered the 'failed' state with result 'exit-code'.
Feb 05 15:48:32 localhost systemd[1]: Failed to start Glorytun TCP on tun0.
░░ Subject: A start job for unit glorytun-tcp@tun0.service has failed

pbix, 2022-02-05:
It seems the installation of omr-glorytun-tcp is failing.  More details below.
=================================================================

root@localhost:#apt-get -V -y -o Dpkg::Options::="--force-confdef" -o Dpkg::Options::="--force-confold" -o Dpkg::Options::="--force-overwrite" install omr-glorytun-tcp=0.0.35-4
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
omr-glorytun-tcp is already the newest version (0.0.35-4).
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up omr-glorytun-tcp (0.0.35-4) ...
Job for glorytun-tcp@tun0.service failed because the control process exited with error code.
See "systemctl status glorytun-tcp@tun0.service" and "journalctl -xe" for details.
dpkg: error processing package omr-glorytun-tcp (--configure):
 installed omr-glorytun-tcp package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 omr-glorytun-tcp
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@localhost:/# systemctl status glorytun
Unit glorytun.service could not be found.
root@localhost:# systemctl status glorytun-tcp
Unit glorytun-tcp.service could not be found.
root@localhost:# systemctl status glorytun-tcp@tun0.service
● glorytun-tcp@tun0.service - Glorytun TCP on tun0
     Loaded: loaded (/lib/systemd/system/glorytun-tcp@.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2022-02-05 16:03:33 UTC; 40s ago
    Process: 19824 ExecStart=/usr/bin/glorytun-tcp-run /etc/glorytun-tcp/tun0 (code=exited, status=1/FAILURE)
    Process: 19825 ExecStartPost=/etc/glorytun-tcp/post.sh /etc/glorytun-tcp/tun0 (code=exited, status=1/FAILURE)
   Main PID: 19824 (code=exited, status=1/FAILURE)
        CPU: 10ms

Feb 05 16:03:33 localhost systemd[1]: glorytun-tcp@tun0.service: Failed with result 'exit-code'.
Feb 05 16:03:33 localhost systemd[1]: Failed to start Glorytun TCP on tun0.
Feb 05 16:03:33 localhost systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 5.
Feb 05 16:03:33 localhost systemd[1]: Stopped Glorytun TCP on tun0.
Feb 05 16:03:33 localhost systemd[1]: glorytun-tcp@tun0.service: Start request repeated too quickly.
Feb 05 16:03:33 localhost systemd[1]: glorytun-tcp@tun0.service: Failed with result 'exit-code'.
Feb 05 16:03:33 localhost systemd[1]: Failed to start Glorytun TCP on tun0.

Ysurac, 2022-02-05:
I'm not able to reproduce the issue on my VPS, I would need to get a VPS on linode.

pbix, 2022-02-06:
If there are any diagnostics steps you can suggest I will be happy to investigate more and report my results.

Ysurac, 2022-02-06:
I can reproduce the issue. I will solve this.

Ysurac, 2022-02-06:
Fixed

pbix, 2022-02-06:
I confirm that the script works again at Linode.  I installed it on Debian 11 successfully.  Thanks for looking at this and for your work on this project.

closed 2022-02-06 by pbix