Ysurac/openmptcprouter #2254: Build Errors at Git Head

pbix, 2022-01-23:
## Expected Behavior

It should be possible to compile this project on Debian 10 by following the instructions here [https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform](url)

## Current Behavior

I know the repo is bleeding edge but thought I would mention that it will not build at the moment when following the instructions on the page.  Build error as follows appears.

/bin/bash ../libtool  --tag=CC   --mode=compile aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I.   -I/home/build/openmptcprouter/rpi4/source/staging_dir/target-aarch64_cortex-a72_musl/usr/include -I/home/build/openmptcprouter/rpi4/source/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/usr/include -I/home/build/openmptcprouter/rpi4/source/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/include/fortify -I/home/build/openmptcprouter/rpi4/source/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/include   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/build/openmptcprouter/rpi4/source/build_dir/target-aarch64_cortex-a72_musl/iperf-ssl/iperf-3.10.1=iperf-3.10.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE  -Wall -MT iperf_tcp.lo -MD -MP -MF .deps/iperf_tcp.Tpo -c -o iperf_tcp.lo iperf_tcp.c
libtool: compile:  aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I/home/build/openmptcprouter/rpi4/source/staging_dir/target-aarch64_cortex-a72_musl/usr/include -I/home/build/openmptcprouter/rpi4/source/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/usr/include -I/home/build/openmptcprouter/rpi4/source/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/include/fortify -I/home/build/openmptcprouter/rpi4/source/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/build/openmptcprouter/rpi4/source/build_dir/target-aarch64_cortex-a72_musl/iperf-ssl/iperf-3.10.1=iperf-3.10.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE -Wall -MT iperf_tcp.lo -MD -MP -MF .deps/iperf_tcp.Tpo -c iperf_tcp.c -o iperf_tcp.o
In file included from iperf_tcp.c:48:
flowlabel.h:41:8: error: redefinition of 'struct in6_flowlabel_req'
 struct in6_flowlabel_req
        ^~~~~~~~~~~~~~~~~
In file included from /home/build/openmptcprouter/rpi4/source/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/include/linux/tcp.h:25,
                 from iperf.h:43,
                 from iperf_tcp.c:41:
/home/build/openmptcprouter/rpi4/source/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/include/linux/in6.h:71:8: note: originally defined here
 struct in6_flowlabel_req {
        ^~~~~~~~~~~~~~~~~
make[6]: *** [Makefile:955: iperf_tcp.lo] Error 1
make[6]: Leaving directory '/home/build/openmptcprouter/rpi4/source/build_dir/target-aarch64_cortex-a72_musl/iperf-ssl/iperf-3.10.1/src'
make[5]: *** [Makefile:711: all] Error 2
make[5]: Leaving directory '/home/build/openmptcprouter/rpi4/source/build_dir/target-aarch64_cortex-a72_musl/iperf-ssl/iperf-3.10.1/src'
make[4]: *** [Makefile:389: all-recursive] Error 1
make[4]: Leaving directory '/home/build/openmptcprouter/rpi4/source/build_dir/target-aarch64_cortex-a72_musl/iperf-ssl/iperf-3.10.1'
make[3]: *** [Makefile:85: /home/build/openmptcprouter/rpi4/source/build_dir/target-aarch64_cortex-a72_musl/iperf-ssl/iperf-3.10.1/.built] Error 2
make[3]: Leaving directory '/home/build/openmptcprouter/feeds/packages/net/iperf3'
time: package/feeds/packages/iperf3/ssl/compile#2.71#0.17#2.84
    ERROR: package/feeds/packages/iperf3 failed to build (build variant: ssl).
make[2]: *** [package/Makefile:116: package/feeds/packages/iperf3/compile] Error 1
make[2]: Leaving directory '/home/build/openmptcprouter/rpi4/source'
make[1]: *** [package/Makefile:110: /home/build/openmptcprouter/rpi4/source/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/build/openmptcprouter/rpi4/source'
make: *** [/home/build/openmptcprouter/rpi4/source/include/toplevel.mk:230: world] Error 2

## Possible Solution

I wonder if a needed tool is not listed for the install.  Possible a minor edit to the documented build procedure might be needed as well or an edit to the makefile.  Not sure really.

## Steps to Reproduce the Problem

  1. Create a fresh Debian 10 install.  In my case I used a Linode VPS and had, after install I had.
      Linux localhost 4.19.0-18-amd64 #1 SMP Debian 4.19.208-1 (2021-09-29) x86_64
  2. Use apt to install the tools listed on the build page.
  3. Follow the build instructions.  Ending with the following command line.
      OMR_TARGET="rpi4" OMR_FEED_SRC="master" ./build.sh -j1 V=s

## Context (Environment)

## Specifications

Built on Linode VPS with Debian 10:

Linux localhost 4.19.0-18-amd64 #1 SMP Debian 4.19.208-1 (2021-09-29) x86_64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

Ysurac, 2022-01-24:
Use develop branch for now or a tagged branch, for now master branch can be not usable as I've not yet merged all in master for next release.

pbix, 2022-01-27:
> Use develop branch for now or a tagged branch, for now master branch can be not usable as I've not yet merged all in master for next release.

The error is the same when building either the "develop" or "master" branch.  Since you are able to make build I suspect something is missing from the tools install instruction.

Ysurac, 2022-01-27:
This is the script used to build images using Github Actions: https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/.github/workflows/main.yml

closed 2022-05-03 by github-actions[bot]