Ysurac/openmptcprouter #2256: Can I use the VPS as a VPN when traveling ?

adamb, 2022-01-25:
## Expected Behavior

I'm running the VPS so my Pi can connect with OMR.  But can I also connect to the VPS as a VPN when I'm out and about? 

If so, how do I do this?

adamb, 2022-01-25:
Great!  I'm in a cafe with my Macbook, how do I connect to my VPS?  What VPN software or settings do I use?

wlatic, 2022-01-26:
You'd need to create a router device running openmptcprouter (USB powered would work) and connected to the cafe wifi, setup to connect to your openmptcp server. There are far easier ways to do what you are looking to do. This software is for combining connects as one single.

adamb, 2022-01-26:
Sorry, I'm not being clear I think.

I want to have OMR running on Pi at my house and connecting to the VPS.  But if I'm not at home and traveling, can I use the same VPS as a VPN _without_ the OMR.  Can my Mac connect to the VPS and use it as a VPN server?  Since I'm paying for the VPS it and OMR is using it as a VPN, it seems like my MacBook should also be able to connect directly to the VPS as a VPN.

The VPS is running Glorytun is a VPN.  So can I connect directly from my Mac to the VPNS using Glorytun?  How do I do that?  OMR is doing that, so I would think that my Mac can do it to...  I would need the Glorytun settings from the VPS I suppose and put that into some client VPN software on the Mac.

Ysurac, 2022-01-26:
VPN allow only one client, so if OMR is running, not possible to use same VPN.
But, wireguard and OpenVPN are also installed on the VPS, and if you don't use "MPTCP over VPN" option, you can use them.
Configurations are in /etc/openvpn and /etc/wireguard on the VPS. 
Wireguard configuration need to be customized, OpenVPN can be used by using certificates.
You can also add your own configuration, you then need to also change /etc/shorewall configuration to allow it.

adamb, 2022-01-26:
Perfect, thanks!

closed 2022-01-26 by adamb