Ysurac/openmptcprouter #2257: Pi-Hole lighttpd fails binding on 10.255.252.1:80

DracoTomes, 2022-01-28:
**Is your enhancement proposal related to a problem? Please describe.**

After installing Pi-Hole on VPS following [this wiki](https://github.com/Ysurac/openmptcprouter/wiki/Pi-hole) the web interface of the Pi-Hole is not reachable. This is because lighttpd fails to start. Using journalctl I got the following error:
`Jan 28 20:13:33 mptcpBTW lighttpd[19310]: 2022-01-28 20:13:32: (network.c.313) can't bind to socket: 10.255.252.1:80 Cannot assign requested address`
Removing `$SERVER["socket"] == "10.255.252.1:80" { }` from /etc/lighttpd/external.conf and lighttpd started correctly.

**Describe the solution you'd like**

Either removing `$SERVER["socket"] == "10.255.252.1:80" { }` from the downloadable script if it's no longer being used or implement a check whether that address is usable.

Adorfer, 2022-01-29:
so just put the littpd on a different port, like 8080 and you are good.

Ysurac, 2022-01-31:
10.255.252.1 is used by OpenVPN TCP, so if you didn't make any changes and use latest VPS release you should have it available and lighttpd should be able to bind to it.