Ysurac/openmptcprouter #2258: BP UL too low compared to DL

GFRT-tech, 2022-01-31:
## Expected Behavior
UL bandwidth approximately equal to DL 82 Mbps

## Current Behavior
On two differents hardware : UL BP is too low 0,24 Mbps max

## Specifications

  - OpenMPTCProuter version: v058.5
  - OpenMPTCProuter VPS version: v0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: MilkyWAN & FirstHeberg
  - OpenMPTCProuter platform: ER-X and AOPEN DE2700

Ysurac, 2022-01-31:
How did you make tests ? What is the UL BP result for each WAN ?

GFRT-tech, 2022-02-01:
test on three different systems  Linux 👍 sin NUC LapTop Desktop :
guyf@guy-pc:~$ iperf3 -c 62.210.156.7 -p 9222 -R
Connecting to host 62.210.156.7, port 9222
Reverse mode, remote host 62.210.156.7 is sending
[  5] local 10.0.1.45 port 47952 connected to 62.210.156.7 port 9222
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  12.4 MBytes   104 Mbits/sec                  
[  5]   1.00-2.00   sec  14.5 MBytes   122 Mbits/sec                  
[  5]   2.00-3.00   sec  14.0 MBytes   117 Mbits/sec                  
[  5]   3.00-4.00   sec  14.6 MBytes   123 Mbits/sec                  
[  5]   4.00-5.00   sec  13.8 MBytes   116 Mbits/sec                  
[  5]   5.00-6.00   sec  13.9 MBytes   117 Mbits/sec                  
[  5]   6.00-7.00   sec  14.5 MBytes   121 Mbits/sec                  
[  5]   7.00-8.00   sec  13.7 MBytes   115 Mbits/sec                  
[  5]   8.00-9.00   sec  14.0 MBytes   117 Mbits/sec                  
[  5]   9.00-10.00  sec  14.3 MBytes   120 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.92  sec   172 MBytes   132 Mbits/sec  107             sender
[  5]   0.00-10.00  sec   140 MBytes   117 Mbits/sec                  receiver

guyf@guy-pc:~$ iperf3 -c 62.210.156.7 -p 9222
Connecting to host 62.210.156.7, port 9222
[  5] local 10.0.1.45 port 47956 connected to 62.210.156.7 port 9222
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  1024 KBytes  8.38 Mbits/sec    1   97.6 KBytes       
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    0   58.0 KBytes       
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    0   58.0 KBytes       
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    0   56.6 KBytes       
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    0   65.0 KBytes       
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    0   65.0 KBytes       
[  5]   6.00-7.00   sec   243 KBytes  1.99 Mbits/sec    0   31.1 KBytes       
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    0   31.1 KBytes       
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   53.7 KBytes       
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    0   53.7 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.24 MBytes  1.04 Mbits/sec    1             sender
[  5]   0.00-10.02  sec   238 KBytes   195 Kbits/sec                  receiver

guyf@guy-pc:~$ iperf3 -t 10 -c scaleway.testdebit.info -p 9222 -4 -R
Connecting to host scaleway.testdebit.info, port 9222
Reverse mode, remote host scaleway.testdebit.info is sending
[  5] local 10.0.1.45 port 47966 connected to 62.210.156.7 port 9222
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  11.9 MBytes  99.5 Mbits/sec                  
[  5]   1.00-2.00   sec  14.1 MBytes   118 Mbits/sec                  
[  5]   2.00-3.00   sec  13.7 MBytes   115 Mbits/sec                  
[  5]   3.00-4.00   sec  14.3 MBytes   120 Mbits/sec                  
[  5]   4.00-5.00   sec  14.4 MBytes   121 Mbits/sec                  
[  5]   5.00-6.00   sec  13.9 MBytes   117 Mbits/sec                  
[  5]   6.00-7.00   sec  14.3 MBytes   120 Mbits/sec                  
[  5]   7.00-8.00   sec  13.3 MBytes   111 Mbits/sec                  
[  5]   8.00-9.00   sec  13.7 MBytes   115 Mbits/sec                  
[  5]   9.00-10.00  sec  14.0 MBytes   117 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.02  sec   163 MBytes   136 Mbits/sec   34             sender
[  5]   0.00-10.00  sec   138 MBytes   115 Mbits/sec                  receiver

guyf@guy-pc:~$ iperf3 -t 10 -c scaleway.testdebit.info -p 9222 -4 
Connecting to host scaleway.testdebit.info, port 9222
[  5] local 10.0.1.45 port 47972 connected to 62.210.156.7 port 9222
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  1.63 MBytes  13.7 Mbits/sec    0   93.3 KBytes       
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    0   93.3 KBytes       
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    0   62.2 KBytes       
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    0   62.2 KBytes       
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    0   28.3 KBytes       
[  5]   5.00-6.00   sec   243 KBytes  1.99 Mbits/sec    0   31.1 KBytes       
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    0   62.2 KBytes       
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    0   62.2 KBytes       
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   62.2 KBytes       
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    0   50.9 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.87 MBytes  1.56 Mbits/sec    0             sender
[  5]   0.00-10.02  sec   408 KBytes   334 Kbits/sec                  receiver

And finaly 👍 
https://www.speedtest.net/fr/result/12695849719

Ysurac, 2022-02-01:
What is 10.0.1.45 that it connect on local ? Your PC ?
Did you try to launch same test on the router directly via SSH ?

GFRT-tech, 2022-02-01:
Since ssh on OpenMPTCProuter 

root@OpenMPTCProuter:~#  iperf3 -c 62.210.156.7 -p 9222 -R
Connecting to host 62.210.156.7, port 9222
Reverse mode, remote host 62.210.156.7 is sending
[  5] local 10.255.255.2 port 54789 connected to 62.210.156.7 port 9222
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  9.68 MBytes  81.2 Mbits/sec
[  5]   1.00-2.00   sec  10.5 MBytes  88.2 Mbits/sec
[  5]   2.00-3.00   sec  14.2 MBytes   119 Mbits/sec
[  5]   3.00-4.00   sec  16.9 MBytes   142 Mbits/sec
[  5]   4.00-5.00   sec  16.4 MBytes   137 Mbits/sec
[  5]   5.00-6.00   sec  16.9 MBytes   142 Mbits/sec
[  5]   6.00-7.00   sec  16.4 MBytes   137 Mbits/sec
[  5]   7.00-8.00   sec  15.6 MBytes   131 Mbits/sec
[  5]   8.00-9.00   sec  16.0 MBytes   134 Mbits/sec
[  5]   9.00-10.00  sec  15.9 MBytes   134 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.02  sec   164 MBytes   137 Mbits/sec  146             sender
[  5]   0.00-10.00  sec   149 MBytes   125 Mbits/sec                  receiver
- - - - - - - - - - - - - - - - - - - - - - - - -
root@OpenMPTCProuter:~# iperf3 -c 62.210.156.7 -p 9222
Connecting to host 62.210.156.7, port 9222
[  5] local 10.255.255.2 port 55169 connected to 62.210.156.7 port 9222
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  2.48 MBytes  20.8 Mbits/sec    0   31.1 KBytes
[  5]   1.00-2.00   sec  1.25 MBytes  10.5 Mbits/sec    4   29.7 KBytes
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    0   29.7 KBytes
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    0   29.7 KBytes
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    1   31.1 KBytes
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    2   31.1 KBytes
[  5]   6.00-7.00   sec  1.25 MBytes  10.5 Mbits/sec    1   28.3 KBytes
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    0   28.3 KBytes
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   28.3 KBytes
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    0   28.3 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  4.98 MBytes  4.18 Mbits/sec    8             sender
[  5]   0.00-10.02  sec   231 KBytes   189 Kbits/sec                  receiver
- - - - - - - - - - - - - - - - - - - - - - - - 
I tested several configurations:
OpenMPTCProuter on ER-X or on AOPEN DE2700 
I tested in ssh directly on the different OpenMPTCProuter with iperf3 no differences the UL BP is still low.
I replaced the ZYXEL USG20VPN router on the LAN output directly with a workstation, then with a laptop, then with a NUC DE3815TYKE.
Whatever the test context I always have a low UL BP.
And the bandwidth differences are very small between the different hardware used for OpenMPTCProuter (Aopen, Edgerouter ER-X).
I also tried to remove the WAN2 to leave only the WAN1, same result.
Translated with www.DeepL.com/Translator (free version)

Behind WAN2 there is no machine

The idea is to pass all the traffic from WAN1
- - - - - - - - - - - - - - - - - - - - - - - - 
[image]
-

Kalimeiro, 2022-02-02:
In my own experience with 2 enterprises using SFR FTTH or with 4G : MPTCP is blocked. (but tcp mptcp capabilities is OK on OMR)

And, you can turn MPTCP over VPN using WireGuard to workaround, but the ONT get multiple disconnect/reconnect when download or upload at high speed.

When not using WireGuard, Download speed Works well, but Upload speed is very very very bad.

GFRT-tech, 2022-02-04:
> In my own experience with 2 enterprises using SFR FTTH or with 4G : MPTCP is blocked. (but tcp mptcp capabilities is OK on OMR)
>
If I understand correctly OpenMPTCProuter will never work with SFR FTTH
Is that correct?

Ysurac, 2022-02-04:
No, it's working. At least on download side. On upload side it can be bad (test yourself) but better when using MPTCP over VPN option.

CrNMGuy, 2022-03-17:
Got same issue (bad upload) with FTTH THD42 in France (collect through is Axione, 200mpbs symetric, Provider is Illyse)

iperf3 upload **without** MPTCP over VPN
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  28.0 MBytes  23.5 Mbits/sec    0             sender
[  5]   0.00-10.00  sec  1.45 MBytes  1.22 Mbits/sec                  receiver

upload **with** MPTCP over VPN
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   131 MBytes   110 Mbits/sec   72             sender
[  5]   0.00-10.00  sec   104 MBytes  87.1 Mbits/sec                  receiver

closed 2022-06-22 by github-actions[bot]