Ysurac/openmptcprouter #2259: Iperf compilation error

moham96, 2022-01-31:
## Expected Behavior
The compilation should complete without an error

## Current Behavior
Compiling OMR following [the wiki](https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform) for turris omnia router results in a compilation error:
```

libtool: compile:  arm-openwrt-linux-muslgnueabi-gcc -DHAVE_CONFIG_H -I. -I/home/alkadm/openmptcprouter/turris_omnia/source/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/include -I/home/alkadm/openmptcprouter/turris_omnia/source/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/include -I/home/alkadm/openmptcprouter/turris_omnia/source/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include/fortify -I/home/alkadm/openmptcprouter/turris_omnia/source/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/alkadm/openmptcprouter/turris_omnia/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/iperf-ssl/iperf-3.10.1=iperf-3.10.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE -Wall -MT iperf_tcp.lo -MD -MP -MF .deps/iperf_tcp.Tpo -c iperf_tcp.c -o iperf_tcp.o
In file included from iperf_tcp.c:48:
flowlabel.h:41:8: error: redefinition of 'struct in6_flowlabel_req'
 struct in6_flowlabel_req
        ^~~~~~~~~~~~~~~~~
In file included from /home/alkadm/openmptcprouter/turris_omnia/source/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include/linux/tcp.h:25,
                 from iperf.h:43,
                 from iperf_tcp.c:41:
/home/alkadm/openmptcprouter/turris_omnia/source/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include/linux/in6.h:71:8: note: originally defined here
 struct in6_flowlabel_req {
        ^~~~~~~~~~~~~~~~~
make[6]: *** [Makefile:955: iperf_tcp.lo] Error 1
make[6]: Leaving directory '/home/alkadm/openmptcprouter/turris_omnia/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/iperf-ssl/iperf-3.10.1/src'
make[5]: *** [Makefile:711: all] Error 2
make[5]: Leaving directory '/home/alkadm/openmptcprouter/turris_omnia/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/iperf-ssl/iperf-3.10.1/src'
make[4]: *** [Makefile:389: all-recursive] Error 1
make[4]: Leaving directory '/home/alkadm/openmptcprouter/turris_omnia/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/iperf-ssl/iperf-3.10.1'
make[3]: *** [Makefile:85: /home/alkadm/openmptcprouter/turris_omnia/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/iperf-ssl/iperf-3.10.1/.built] Error 2
make[3]: Leaving directory '/home/alkadm/openmptcprouter/feeds/packages/net/iperf3'
time: package/feeds/packages/iperf3/ssl/compile#4.33#0.58#7.76
    ERROR: package/feeds/packages/iperf3 failed to build (build variant: ssl).
make[2]: *** [package/Makefile:116: package/feeds/packages/iperf3/compile] Error 1
make[2]: Leaving directory '/home/alkadm/openmptcprouter/turris_omnia/source'
make[1]: *** [package/Makefile:110: /home/alkadm/openmptcprouter/turris_omnia/source/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/alkadm/openmptcprouter/turris_omnia/source'
make: *** [/home/alkadm/openmptcprouter/turris_omnia/source/include/toplevel.mk:230: world] Error 2

```
## Specifications

  - OpenMPTCProuter version: master branch
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: Turris omnia router

Ysurac, 2022-01-31:
Clone of #2254
The iperf3 package used here is not the right one, this use openwrt package and OpenMPTCProuter package have the needed patch.

closed 2022-01-31 by Ysurac

moham96, 2022-01-31:
@Ysurac How to tell the build script to use the OpenMPTCProuter package? I'm following the exact steps in the wiki.

Ysurac, 2022-01-31:
For now you may need to use a tagged branch, like "0.58.5" as I'm pushing change for next release on master branch.

moham96, 2022-01-31:
@Ysurac using a tagged branch doesn't really work for me, it stops when trying to switch branches to the `build` branch
```

Switched to a new branch 'build'
Reset branch 'build'
error: Your local changes to the following files would be overwritten by checkout:
        luci.mk
Please commit your changes or stash them before you switch branches.
Aborting
```

Ysurac, 2022-01-31:
There is no "build" branch, and yes you need to remove feeds directory if already exist.