Ysurac/openmptcprouter #2261: VPS Multiple IP's

lirees, 2022-02-02:
## Expected Behavior

Using Multiple IP's Adddress form VPS (  https://github.com/Ysurac/openmptcprouter/wiki/VPS-Multi-IPs ) 

## Current Behavior

the script are run after the addition ip's configuration on the VPS 

i can reach each pubblici IP's  but from the orm i can't forward the traffic on lan from Services->Shadowsocks-libev->Redir Rules 

i try to add a new rules as show in the howto 
label : rule fof my ip1
ss-redir for TCP : hi1- tcp_and_udp
Local-out default : bypass
Src ip/net forward : my_lan_ip ( 192.168.100.1) 

i click save but nothing happen

## Specifications

  - OpenMPTCProuter version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: Aruba
  - OpenMPTCProuter platform: ESXI

lirees, 2022-02-02:
I started from scratch and now and now in the Services->Shadowsocks-libev->Redir Rules i see oip0_rule and oip1_rule now if i 

but now i have a problem with a port forwarding
- if i try to forward a port from vpn zone to a lan ip address all it's ok from all the pubblic ip's 
- if i specify from which public IP i want forward the traffic ( Firewall - Port Forwards - Advanced Settings > External IP address :  gre4-oip0gre ) to internal lan ip the ORM becomes unreachable

thanks

Ysurac, 2022-02-04:
External IP address must be specified manually

kekemar, 2022-02-13:
Hi,

I have same problem, i have 2 ip in my VPS : 

### Config files
auto eth0
iface eth0 inet dhcp
    mtu 1500

auto eth0:1
iface eth0:1 inet static
    address "second IP"
    netmask 255.255.255.255

### Redirection
If i redirect a port in firewall, example 80, the first and the second ip public is redirected.
If i specify the first ip public to the external option, don't redirect
If i specify the second ip public to the external option,  only the second ip is redirected
If i specify the first ip public to the external option, to redirect to an internal ip at port 80 and specify the second ip public to the external option to redirect to other internal ip at port 80, it don't work

Thanks

closed 2022-05-19 by github-actions[bot]

dcousins21, 2024-11-13:
Hi, 

I have the same issue as above. 

I have 2 external IP's on my VPS. 

If I setup a port forward and leave external IP as "any" - the traffic passes through from either of the external IPs to the specified lan IP. If i specify one of the 2 external IP's nothing gets through to the specified LAN address. 

Did anyone solve the above issue?

Thanks