Ysurac/openmptcprouter #2262: Lots of issues after removing 1 interface

finnschi, 2022-02-04:
So my contract on 1 of my DSL lines ended.

after the contract period was up they started syncronising my router at like 480kbps -
When this happend weird things started to happen, some images on websites wouldnt load, discord messages took forever to send , weird ping spikes to 8.8.8.8. 

So allright Ill jsut remove that interface.. done, restarted everything.. (router and VPS) 

Now I am not able to ping 8.8.8.8 anymore but some websites still works, which I think has something to do with ipv6/v4  but then for some seconds I can ping it again.  Ping to my VPS is running constantly without problems 

There also seems to be a issue with V2Ray all of the sudden.

What are my next moves here? reinstall vps? reinstall router?  Dont know where to start , everything was working just fine before I remove the dsl line.  DSL line was NOT my multipath master. 

Right now in status I see:
Router:
No IPv6 Access
VPS:
No Output
No Server IP adress
No Public IP adress
Cant get Public Ip from V2Ray

But them after some minutes it goes all fine but then goes back to this error, sometimes it only shows V2ray not running. 
And then sometimes the router interface just murders itself with some 
TypeError
undefined is not an object (evaluating 'boardinfo.release.description')

Logs from the last 10minutes, as you can see its getting a bit *crazy* 

Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.4.138.53 port 53
Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.134.53 port 53
Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.134.53 port 53
Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.4.138.53 port 53
Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.4.138.53 port 53
Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.135.53 port 53
Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.135.53 port 53
Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.133.53 port 53
Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.133.53 port 53
Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.134.53 port 53
Fri Feb  4 16:21:59 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.134.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.133.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.133.53 port 53
Fri Feb  4 16:22:00 2022 daemon.info dnsmasq-dhcp[14679]: DHCPRELEASE(eth0) 192.168.33.205 8c:85:80:a1:a1:52
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.135.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.135.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.134.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.134.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.133.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.133.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.135.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.135.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.133.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.133.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.134.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.134.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.133.53 port 53
Fri Feb  4 16:22:00 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.133.53 port 53
Fri Feb  4 16:22:01 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.135.53 port 53
Fri Feb  4 16:22:01 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.135.53 port 53
Fri Feb  4 16:22:01 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.134.53 port 53
Fri Feb  4 16:22:01 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.134.53 port 53
Fri Feb  4 16:22:01 2022 user.notice omr-bypass: Restart dnsmasq...
Fri Feb  4 16:22:01 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.135.53 port 53
Fri Feb  4 16:22:01 2022 daemon.err unbound: [18658:0] error: read (in tcp s): Connection refused for 199.43.135.53 port 53
Fri Feb  4 16:22:01 2022 user.notice omr-bypass: OMR-ByPass is running
Fri Feb  4 16:22:02 2022 daemon.info v2ray: DNS disabled: main_dns
Fri Feb  4 16:22:02 2022 daemon.err omr-tracker[19722]: [info] DNS disabled: main_dns
Fri Feb  4 16:22:02 2022 daemon.info v2ray: Setting transparent proxy on port: 1897
Fri Feb  4 16:22:02 2022 daemon.err omr-tracker[19722]: [info] Setting transparent proxy on port: 1897
Fri Feb  4 16:22:02 2022 daemon.info v2ray: Transparent proxy mode: default
[... 583 lines cut ...]
Fri Feb  4 16:28:54 2022 user.notice ModemManager: hotplug: add network interface tun0: event processed
Fri Feb  4 16:28:54 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Fri Feb  4 16:28:54 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb  4 16:28:54 2022 user.notice SQM: Starting SQM script: piece_of_cake.qos on tun0, in: 0 Kbps, out: 0 Kbps
Fri Feb  4 16:28:54 2022 user.notice SQM: piece_of_cake.qos was started on tun0 successfully
Fri Feb  4 16:28:54 2022 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Fri Feb  4 16:28:54 2022 user.notice ModemManager: hotplug: add network interface ifb4tun0: event processed
Fri Feb  4 16:28:54 2022 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4tun0, subsystem=net
Fri Feb  4 16:28:54 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb  4 16:28:54 2022 user.notice post-tracking-post-tracking: Set firewall on server
Fri Feb  4 16:28:55 2022 daemon.err glorytun[22449]: 202.61.249.125.65001: connected
Fri Feb  4 16:28:55 2022 daemon.info glorytun[22449]: STARTED tun0
Fri Feb  4 16:28:56 2022 daemon.debug dnsmasq[26518]: listening on tun0(#31): 10.255.255.2 port 53
Fri Feb  4 16:28:56 2022 user.notice omr-bypass: Starting OMR-ByPass...
Fri Feb  4 16:29:03 2022 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Fri Feb  4 16:29:07 2022 daemon.info dnsmasq-dhcp[26518]: DHCPDISCOVER(eth0) c0:c9:e3:a3:30:34
Fri Feb  4 16:29:07 2022 daemon.info dnsmasq-dhcp[26518]: DHCPOFFER(eth0) 192.168.33.237 c0:c9:e3:a3:30:34
Fri Feb  4 16:29:10 2022 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Fri Feb  4 16:29:22 2022 user.notice omr-bypass: Reload dnsmasq...
Fri Feb  4 16:29:22 2022 daemon.info dnsmasq[26518]: read /etc/hosts - 4 addresses
Fri Feb  4 16:29:22 2022 daemon.info dnsmasq[26518]: read /tmp/hosts/dhcp.cfg01411c - 5 addresses
Fri Feb  4 16:29:22 2022 daemon.info dnsmasq-dhcp[26518]: read /etc/ethers - 0 addresses
Fri Feb  4 16:29:22 2022 user.notice omr-bypass: OMR-ByPass is running
Fri Feb  4 16:30:03 2022 daemon.info dnsmasq-dhcp[26518]: DHCPDISCOVER(eth0) c0:c9:e3:a3:30:34
Fri Feb  4 16:30:03 2022 daemon.info dnsmasq-dhcp[26518]: DHCPOFFER(eth0) 192.168.33.237 c0:c9:e3:a3:30:34

Ysurac, 2022-02-04:
Please use the template.
Also check that you have an interface set as master.

finnschi, 2022-02-04:
what template? I have 1 interface set as master.

Ysurac, 2022-02-04:
These templates you get when opening an issue: https://github.com/Ysurac/openmptcprouter/issues/new/choose

I would need the config from System->OpenMPTCProuter->Show all settings

finnschi, 2022-02-04:
Sorry they didnt pop up for some reason, will make sure next time!!  Hope I removed anything sensitive 

`
adblock.global=adblock
adblock.global.adb_debug='0'
adblock.global.adb_safesearch='0'
adblock.global.adb_dnsfilereset='0'
adblock.global.adb_mail='0'
adblock.global.adb_report='0'
adblock.global.adb_backup='1'
adblock.global.adb_dns='unbound'
adblock.global.adb_fetchutil='curl'
adblock.global.adb_backupdir='/etc/adblock'
adblock.global.adb_maxqueue='16'
adblock.global.adb_sources='adaway' 'adguard' 'adguard_tracking' 'android_tracking' 'andryou' 'anti_ad' 'anudeep' 'disconnect' 'energized' 'firetv_tracking' 'games_tracking' 'notracking' 'oisd_full' 'reg_de' 'smarttv_tracking' 'stopforumspam' 'winspy' 'yoyo'
adblock.global.adb_forcedns='0'
adblock.global.adb_trigger='wan5'
adblock.global.adb_enabled='0'
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].dnssec='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].server='45.90.28.81' '45.90.30.81'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/zoom.us/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/goodbyekansas.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/ifconfig.me/omr_dst_bypass_all,omr6_dst_bypass_all' '/uploaded.to/omr_dst_bypass_all,omr6_dst_bypass_all' '/ul.to/omr_dst_bypass_all,omr6_dst_bypass_all' '/time.windows.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/paypal.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/netflix.com/omr_dst_bypass_all,omr6_dst_bypass_all,omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxext.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxvideo.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxso.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/fast.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/dmqdd6hw24ucf.cloudfront.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/d25xi40x97liuc.cloudfront.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/aiv-delivery.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/aiv-cdn.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/1s3.lvlt.dash.us.aiv-cdn.net.c.footprint.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/s.loris.llnwd.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/atv-ext.amazon.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/atv-ps.amazon.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/media-amazon.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/amazonvideo.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/primevideo.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/cloudfront.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/llnw.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/spotify.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/scdn.co/omr_dst_bypass_all,omr6_dst_bypass_all' '/pscdn.co/omr_dst_bypass_all,omr6_dst_bypass_all' '/spotifycdn.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/spotifycdn.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/audio4-ak-spotify-com.akamaized.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/audio-ak-spotify-com.akamaized.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/heads-ak-spotify-com.akamaized.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/spotify-com.akamaized.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/spotify.com.edgesuite.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/spotify.map.fastly.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/spotify.edgekey.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/spotify.demdex.net/omr_d [... cut ...]
v2ray.omrrouting=routing_rule
v2ray.omrrouting.type='field'
v2ray.omrrouting.inbound_tag='omrbridge'
v2ray.omrrouting.outbound_tag='omrout_tunnel'
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth0' 'eth1' 'eth2' 'eth3' 'eth5'

`

Ysurac, 2022-02-04:
I don't see issue here.
What is the result if you start try with only one interface ?

finnschi, 2022-02-04:
I really do not get it, I reinstalled (factory reset)  the router with latest beta also.. [openmptcprouter v0.58.5 r0+16328-f441be3921]

no V2Ray connection possible , I tried with 1 interface even. 

Gloryturn works of course but I am just getting "Cant Get Public IP adress from V2RAY  and still CANT contact Server Admin Script ...

finnschi, 2022-02-04:
OK so what fixed it for me: 

-Reinstalled both router and VPS from scratch using latest STABLE build. 

V2Ray just went up and seems to be saying up , no errors all is well. 

I have 0 ideas why removing a connection caused this but hey.. I am happy

closed 2022-02-04 by finnschi

Adorfer, 2022-02-05:
> I have 0 ideas why removing a connection caused this but hey.. I am happy

I saw this in the past too, since that i just disable interfaces for MPTCP, i never remove one.