Ysurac/openmptcprouter #2263: Seamless failover of master connection on 2xLTE

highfreq, 2022-02-05:
## Expected Behavior

Connection stays up if one of the LTE links goes down 

## Current Behavior

If Master goes down connection is lost for 15-20 seconds. Pinging tunnel from VPS times out for 15-20 seconds as soon as master link is disconnected.

## Specifications

2 x LTE modems
OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
OpenMPTCProuter VPS provider: Aruba
OpenMPTCProuter platform: RPi4

Ysurac, 2022-02-06:
Should be better in latest OpenMPTCProuter beta/snapshot.
Only one master must be defined.

highfreq, 2022-02-06:
I defined only 1 master, is latest snapshot stable enough to be used in production?

Ysurac, 2022-02-06:
Snapshot/beta is not yet ready for production, it still need more tests.

highfreq, 2022-02-06:
when do you think it'll be out? more or less?

Ysurac, 2022-02-06:
No idea, I hope soon...

highfreq, 2022-02-06:
Ok, thanks :)

highfreq, 2022-02-07:
Is multiple router on one VPS something that will be possible on next release?

Ysurac, 2022-02-07:
No.

closed 2022-05-15 by github-actions[bot]

TalalMash, 2022-05-21:
Seamless failover is working with openmptcprouter v0.59beta6-5.4 r0+16591-b4ea8e1089
A milestone passed :)