Ysurac/openmptcprouter #2265: the site at 192.168.100.1 seems to be unavailable

weasel18, 2022-02-07:
## Expected Behavior
access web-ui of new openmptcprouter on my racknerd VPS

## Current Behavior
using vnc and a browser, going to 192.168.100.1 it tries forever to load then eventually returns "The site at http://192.168.100.1 seems to be unavailable. " and if I attempt to ping or ssh into the address from the VPS it just times out

I've gone through the whole install process multiple times, using Debian and tried it with ubuntu too and nothing has allowed me to access the web interface of openmptcp

## Specifications

  - OpenMPTCProuter version: --
  - OpenMPTCProuter VPS version: 0.1026
  - OpenMPTCProuter VPS provider: RackNerd KVM 2.5gb ram
  - OpenMPTCProuter platform: x86_64 debian 10

Ysurac, 2022-02-07:
OpenMPTCProuter have 2 side, VPS side and router side, only router side have a web interface. Did you install and configure router side ?

weasel18, 2022-02-07:
> OpenMPTCProuter have 2 side, VPS side and router side, only router side have a web interface. Did you install and configure router side ?

OK. Maybe thats where I went wrong. I thought the vps would need to be logged into to configure.

closed 2022-02-24 by weasel18