Ysurac/openmptcprouter #2266: MPTCP Blocked connection as Backup?

chuotde, 2022-02-10:
Hi there,

I have a very high latency WAN connection (actually it is a satellite link) and it is a MPTCP filtered link (already verified with MPTCP check tool). Can I set that connection as a Backup? I want OMR to use that connection only when all other connections go down, I don't want to bond that connection.

Please advise.
Thanks

Kalimeiro, 2022-02-10:
you need to use MPTCP over VPN on this link and set OMRVPN has backup in MPTCP settings (and not the WAN of this link).

chuotde, 2022-02-11:
Hi Kalimeiro,

Thanks for your input. But my idea is if that link is the only link available, is there any way to bypass the OMR and let traffic go through the link directly to Internet rather than to VPS and to Internet? It may then not add up some latency to users.

Cheers

Kalimeiro, 2022-02-11:
Hi,

the subject of you'r second message is totally different from you'r first message.

You say you want to use your link has a backup connection, that suppose you have other wan.

And in the second message tell you have only one wan and don't want to use them rather than the VPS and by-pass directly to internet, so : don't set a VPS on you'r OMR and thats good, or switch directly to OpenWRT instead OMR.

Great day.

closed 2022-05-17 by github-actions[bot]