Ysurac/openmptcprouter #2267: Bonding support

sadahsan, 2022-02-11:
I have two 1G interfaces (eth0 and eth1), I created a bond through Network -> interfaces.

But that Bond interface is not visible under System -> OpenMPTCProuter which means I am unable to assign that bond interface to LAN and WAN (with VLANs), is this supported?

Thanks

Ysurac, 2022-02-11:
Bond interfaces are not working well with MPTCP, but you can still try and set interfaces in Network->Interfaces with multipath enabled, but maybe better to dedicate an interface to lan and one to WANs.

natzakaria, 2022-02-27:
sadahsan - have you managed to get the bonding working?

sadahsan, 2022-02-27:
I managed to setup the bond interface but in the end didn't use it for WAN links.

According to Ysurac bond doesn't seems to work well with Mptcp so I left it there.

closed 2022-06-03 by github-actions[bot]