Ysurac/openmptcprouter #2269: Omr-bypass Netflix not working on smart tv

AlexGiroud, 2022-02-11:
## Expected Behavior
Netflix working on the LG smart tv.

## Current Behavior

Can't reach Netflix servers on the smart tv, the tv got internet and is setup with the good DNS (openmptcprouteur IP) .

The bypass seems to work fine on the phones/PCs in the house

Is there a specific IP to add to the bypass list for LG smart tvs ? 
(I already have the Netflix protocol and these address in the bypass list https://openconnect.netflix.com/mobiledeliverydomains.txt )

AlexGiroud, 2022-02-11:
[image]

vinceducat, 2022-02-12:
Hello as @Kalimeiro said you can use the by-pass.

what is you VPS provider, some time Netflix block the vps IP.

with Milkywan French provider no problem to use Netflix without ByPass ;-)

jamesbrindle, 2022-02-12:
If you're using your router as a DHCP server, and your TV gets its IP address from the OpenMPTCP router, then I've found that you can can add the IP address to the Source LAN IP address or network... Unfortunately for me, I don't want to use the OpenMPTCP router as the DHCP server. Instead I use another router for that, and the OpenMPTCP router is just for WAN (internet), so I'm trying to figure out a way around it. I tried just plugging my TV only into the OpenMPTCP router, but then plex and media on my PC won't work!  Bloody annoying.

Kalimeiro, 2022-02-12:
Netflix and other TV Provider block access of largest datacenter like OVH (by filtering ASN probably ?)

Milkywan works, but it's just a time question to Netflix block this provider.

The solution is to set in omr-bypass an ASN bypass rule, and it works.

jamesbrindle, 2022-02-12:
Hey,

Thanks for that...

I do have a ASN bypass rule, but it isn't working.... I don't suppose you
know which rules to apply?

Regards,

J

AlexGiroud, 2022-02-15:
> Hello as @Kalimeiro said you can use the by-pass.
> 
> what is you VPS provider, some time Netflix block the vps IP.
> 
> with Milkywan French provider no problem to use Netflix without ByPass ;-)

I'm using Ikoula as my VPS provider, but i think that i'll move the VPS side to my appartement in the city (i have a really good fiber connexion with a dedicated ipv4) to solve all these problems in the future.

> Netflix and other TV Provider block access of largest datacenter like OVH (by filtering ASN probably ?)
> 
> Milkywan works, but it's just a time question to Netflix block this provider.
> 
> The solution is to set in omr-bypass an ASN bypass rule, and it works.

I've just tried to filter netflix by ASN, now it's working nowhere...

closed 2022-02-15 by AlexGiroud