Ysurac/openmptcprouter #2270: Optimal requirements

sonnyxp2008, 2022-02-12:
Hi everyone, I wanted to know the optimal requirements for a performing vpn. I currently use 1vcore and 3gb of ram. To have excellent performance is it enough or better to add 1 more vcore or more ram? Thank you

zsoos, 2022-02-16:
I am using vultr.com with 1 core and 1 GB ram. Supports about 650/35 Mbps without a hitch -- a result of ISP 400/20 and Starlink service, about 150/15.

Adorfer, 2022-02-16:
what do you call "optimal"? 
my tests ended with VPS with true 2.5GBit/s connectiviy. there the barrier is about 2GBit/s with dedicated vcores. 
But certainly there will be people here with 40GBit/s, at least as a test run.

sonnyxp2008, 2022-02-16:
I'll explain better. I noticed that when the VPS (dedicated 1vcpu 3gb ram) is put under pressure by adding several push stations, it starts to provide low performance. I don't understand if it depends on the Vcpu or the ram.
I exclude my hardware because OMR is installed on a xeon with 32 gb of ram and 6 gigabit ports. I tried with a smaller pc but the result doesn't change. I also tried the various congestion settings. Speedtest on the VPS side always gives me about 900 mb down and up. The problem lies only when it is put under strain that it starts to drop some packets or the ping becomes unstable.

closed 2022-05-22 by github-actions[bot]