Ysurac/openmptcprouter #2271: no IP address after VPS install on private Virtual machine

primerump, 2022-02-15:
Trying to install OMR server on private VM behind firewall/router, this works if the target is a "bare metal" device, I have an intel I5  mini-pc working right now. If I install a basic debian 10 server on a VM either in proxmox or ESXi and run the server vps script, it installs without error but the IP address ( IPv4 ) is absent.. this only seems to happen with with a  VM..
Is this intended or is there a real issue here ?

thanks

Ysurac, 2022-02-15:
It's not intended, and VPS are often proxmox or ESXi, so I think there is something not classic on your VM install.

Ysurac, 2022-02-15:
For default VPS interface, you can set "INTERFACE=eth0" (replace eth0 by what you want).
But in any case IP should not be removed, a firewall is installed but ip should stay as it's not configured by iptables.

Jimmy062006, 2022-02-15:
My Debian 10 Proxmox VM that used to work also does this on a clean build.

The network adapter changes from ens18 to enpxxxx after running the install script and restarting to the new kernel

Jimmy062006, 2022-02-16:
An update to this the interface line in the script DOSE find the correct interface name.

root@rtr-home:~# echo ${INTERFACE:-$(ip -o -4 route show to default | grep -m 1 -Po '(?<=dev )(\S+)' | tr -d "\n")}
ens18

Jimmy062006, 2022-02-16:
Temp fix is to make the following changes.

Edit `/etc/shorewall/params.net` and `/etc/shorewall6/params.net`  with your new interface name

Then configure your /etc/network/interfaces to setup the new network card.

```# The primary network interface
auto enp6s18
iface enp6s18 inet static
        address 8x.xxx.xxx.xxx
        netmask 255.255.255.240
        gateway 8x.xxx.xxx.xxx
        dns-nameservers 8x.xxx.xxx.xxx```

Reboot and the VPS comes up as expected but nothing else works as expected.

Jimmy062006, 2022-02-16:
Although nothing functions as you'd expect so its kinda useless now. :(

primerump, 2022-02-16:
Well I did find something weird.. if i install debian into a "Q35" based VM it fails with the enpxxx interface as described in my first post.
If I install into an "i440fx" based VM then the install completes correctly and the interface is up with a valid IP ( IPv4 ) address even after reboot..
very odd.. so I am using this as I type.. I don't know the reason for the failure and probably never will.. !! LOL.
thank you Yannick for your prompt reply and excellent work..

closed 2022-05-22 by github-actions[bot]