Ysurac/openmptcprouter #2272: OMR and Samba AD

vinceducat, 2022-02-15:
## have OMR working with machine acting as SAMBA-AD (active directoy)

## At this time taking information to configure the two machine OMR and SANBA AD

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version:  0.1026
  - OpenMPTCProuter VPS provider: MilkyWan
  - OpenMPTCProuter platform: x86_67

need the was to configure OMR because in thos case its Samba actif as DNS DHCP NTP server with this machine have fixe ip etc etc...

so how to configure OMR in this case

Ysurac, 2022-02-15:
You should check OpenWrt doc for that, it's not directly related to OpenMPTCProuter and AD is not something I can help as I have no windows computer and I don't use AD.

vinceducat, 2022-02-15:
could be a start of solutionhttps://secretweaponsdigital.wordpress.com/2016/11/17/admin-small-network-with-openwrt-and-windows-domain-controller/

Adorfer, 2022-02-16:
AD works perfect over OMR. there are no blocking issues. 
(working here formerly with WS2008R2, WS2012, WS2016 now.)

vinceducat, 2022-02-16:
@Adorfer thanks.

is the windows ad server act as DHCP NTP DNS server?
in this case what was the configuration of OMR ?

Adorfer, 2022-02-17:
> @Adorfer thanks.
> 
> is the windows ad server act as DHCP NTP DNS server? in this case what was the configuration of OMR ?

The AD does just AD. No DHCP, no NTP, only split DNS over this strange MS-Protocoll. 

But perhaps i do not get you question what do you want to do? What do you mean mean "working with"? Internet access works for the AD controller, correct?

vinceducat, 2022-02-17:
in most of the tuto on AD, it is recommand to set AD act as DHCP DNS and NTP, so that why i have all this question ...

but perhaps i wrong . because i have understand that the role of DNS DHCP and NTP must be done by AD server...

closed 2022-05-25 by github-actions[bot]