Ysurac/openmptcprouter #2273: Request for configuration for streaming services

zsoos, 2022-02-16:
## Expected Behavior

With out-of-the-box configuration streaming services are not working, complain about VPN/Proxy use and are asking to disconnect. Please create a tutorial how to configure to support streaming services.

Please support the streaming services I am using (USA):

- Netflix (working, can provide details based on discussions here)
- Youtube TV (working, not sure why)
- Hulu (not working)
- Amazon Prime Video (not working)
- HBO MAX (not working)

Other popular streaming services (equally important :-) :

- Showtime Anytime
- Peacock TV
- Apple TV
- Disney+
- etc...

With other words, it would be nice to have a universal solution to support streaming. 

## Current Behavior

This wan binding solution is unique and great work. Streaming services support will be more important as time goes by. Lack of streaming support might be a deal breaker for some.

## Specifications

  - OpenMPTCProuter version:  5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS version:  OpenMPTCProuter VPS 0.1026
  - OpenMPTCProuter VPS provider: VULTR.com
  - OpenMPTCProuter platform:  Qotom Mini PC 8GB RAM 64GB SSD 4 Intel LAN Ports x86_64

Adorfer, 2022-02-16:
as with any other topic like this in the past: it's a problem with the VPS IP you are using. 
if you want streaming with MPTCP you have to go to a VPS hoster which offers IPs which are not in "proxy blacklists". 
(And yes, those tend to cost more.)

zsoos, 2022-02-16:
Do you have any recommendation for such a hosting service in the Los Angeles area?

zsoos, 2022-02-17:
The ISP that I have selected for the VPN (vultr.com) is just fine, my server's IP is green on any of the blacklist checkers. The problem is that the streaming services are detecting the existence of a VPN server and automatically block the connection. I guess they are automatically assume that the VPS is used for circumventing the geographical restrictions. They are not smart enough to detect that my two wan feeds are in the same region (USA).

Looking at previous comments/discussions I figured out how to configure the openmptcprouter to enable the services. There are two important steps:

- OpenMPTCProuter IP must be used as DNS. You do that by configuring your internal router serving local devices (in my case 192.168.32.1 that is changed from the default x.x.100.1). 
- Configure Services/OMR-Bypass on your openmptcprouter. I  guess OMR stands for openmptcprouter.

For streaming services you want to bypass multipath and assign the service to one of the WAN interfaces (in my case Spectrum). The key here is the ASN number for the particular streaming service. Find out more at https://afrinic.net/asn. This way you loose redundancy/bandwidth provided by multipath, but the service works at least.

Here is the configuration for my case:

[image]

Here are a few ASN numbers -- but your best friend is google here:

AS40027	Netflix	
AS23286	Hulu	
AS14618	HBOMAX	
AS16509	Amazon	
AS15169	Youtube

In addition I have also configured the "Domains" section, but to be hones I am not sure if it is needed at all. 
[image]

I hope this helps to clarify things :-)

primerump, 2022-02-19:
streaming from third party VPN is always a cat annd mouse game. most content streamers don't actually scan for a vpn, they just assume and block data center IP address blocks as they appear.
I have a vps in houston,tx and if i use that as my omr endpoint, all those services you mention cry and inform me that im using a vpn. If i use my daughters symetrical fiber service at her home as my omr endpoint, guess what .... no complaints at all..!

zsoos, 2022-02-20:
Thank you @primerump and @Adorfer for your comments. After fighting with the domains and IPs with the vultr server I finally took the plunge and hosted the VPN server on Google Cloud. Using the smallest instance (1 vprocessor, 640K ram) and it seems to be working right now, all the streaming services are up and happy. Not sure about the cost of bandwidth, but the machine runs for about $6/month. I was thinking about hosting a Raspberry Pi as a VPN server at my friends fiber service near by, but that also comes with it's complications. @primerump you are lucky to have a family connection to great bandwidth :-)

Ysurac, 2022-02-20:
Only x86_64 is supported on VPS side for now. So this will not work on a RPI (or you need to compile a RPI4 5.4 kernel with MPTCP patches).

zsoos, 2022-02-20:
Above my grade level :-) Regardless, appreciate the work you are doing @Ysurac :-)

fxousselin, 2022-03-28:
I am currently in France. I use OVH as VPS/Provider. Their first offer is called OverTheBox, which is a packaged version of OMR (https://www.ovhtelecom.fr/overthebox) . It comes with a private IP seen as a xDsl private Ip. and a pre-configured box. It work seamlessly with all french streaming providers.
When I use OMR with one of their public VPS, it’s blocked by Netflix and other streaming platforms. Here is the reason :

**WithOvertTheBox  :** 
inetnum:        151.127.0.0 - 151.127.63.255
netname:        OVH-DSL
org:            ORG-OT26-RIPE
descr:          **DSL static IP**
descr:          www.ovh.com
country:        FR
admin-c:        OK217-RIPE
tech-c:         OTC2-RIPE
status:         LEGACY
mnt-by:         OVH-MNT
created:        2016-09-30T17:16:21Z
last-modified:  2016-09-30T17:16:21Z
source:         RIPE

**With the VPS :** 
inetnum:        146.59.225.0 - 146.59.225.255
netname:        VPS-GRA8
country:        FR
org:            ORG-OS3-RIPE
**geoloc**:         50.98721 2.120542
admin-c:        OTC2-RIPE
tech-c:         OTC2-RIPE
**status**:         LEGACY
mnt-by:         OVH-MNT
created:        2020-11-03T08:58:17Z
last-modified:  2020-11-03T08:58:17Z
source:         RIPE

Conclusion : if you don’t have your « own » public Ip at the end of ORM, you are detected by streaming services as « anonymous » and then blocked. You have to find a provider where you are not mixed with « everybody else ». Hope it helps.
Regards

TalalMash, 2022-04-10:
Oracle VCN blocks are not blacklisted yet, they have free-forever tier too.

closed 2022-08-07 by github-actions[bot]