Ysurac/openmptcprouter #2275: VPS in Cloud - which ports are required to be opened for inbound connections

sadahsan, 2022-02-18:
Hi,

I would appreciate your advise that which ports (TCP, UDP, ICMP) should I open for the inbound connection on VPS running in cloud?

Thanks

sadahsan, 2022-02-18:
Thank you, much appreciated.

closed 2022-02-18 by Ysurac

sadahsan, 2022-02-18:
I am planning to disable below services on VPS as I won't be using them, do see any problem with it?

dsvpn-server@dsvpn0.service
glorytun-udp@tun0.service
mlvpn@mlvpn0.service
openvpn@bonding1.service
openvpn@bonding2.service
openvpn@bonding3.service
openvpn@bonding4.service
openvpn@bonding5.service
openvpn@bonding6.service
openvpn@bonding7.service
openvpn@bonding8.service
openvpn@tun0.service
openvpn@tun1.service
wg-quick@wg0.service

Thanks

sadahsan, 2022-02-18:
Thank you Sir.

sadahsan, 2022-02-18:
Hi again,

I am following your instruction as below and it seems like UDP is still going via glorytun.

[image]

[image]

Ysurac, 2022-02-18:
advanced settings checkbox in wizard, not the tab.