Ysurac/openmptcprouter #2277: Routing does not work when using SRT

kzzanuda, 2022-02-20:
## Expected Behavior

Hi! Unfortunately, packet routing does not work when broadcasting over the SRT protocol. All the configuration recommendations for working with UDP were used, but this does not help.

## Current Behavior

When working with the Internet normally, it is clear that all 4 channels are used as intended. In the broadcast mode, the video stream ceases to be divided and works only on 1 channel. Tell me, what else could be the problem?
I used V2Ray with Glorytun UDP.

## Specifications

  - OpenMPTCProuter version: [v0.58.5](https://github.com/Ysurac/openmptcprouter/releases/tag/v0.58.5)
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: RPI3 B+
  -

aircooled76, 2022-02-21:
I am interested in this topic.  Were you able to send a "normal" video stream like an RTMP stream correctly?

kzzanuda, 2022-02-24:
> I am interested in this topic.  Were you able to send a "normal" video stream like an RTMP stream correctly?

I'm try send video by RTMP, but it's working every bit as.

closed 2022-05-30 by github-actions[bot]