Ysurac/openmptcprouter #2282: V2Ray Not Running / Dropping Out - RPI4 - 0.58.5

jamesbrindle, 2022-02-23:
## Expected Behavior

Switch to V2Ray as the default proxy and it works - and be stable.

## Current Behavior

Either it never comes up (it just says 'V2Ray is not running') or it eventually comes up, and everything is OK for about 5 minutes, then it goes down again.

## Steps to Reproduce the Problem

On the luci interface, go to: 

`System -> OpenMTPCProuter -> (tab) Settings Wizard -> (check) -> Show advanced settings:`

Then switch 'Defaulf proxy' to 'V2Ray'.

Wait a good amount of time until everything has reloaded and see in the OpenMPTCProuter 'Status' section starts and you get a green 'tick' on both the VPS and OpenMPTCProuter. 

If after a good amount of time V2Ray is still not up and running, then perform a reboot and repeat.

## Context (Environment)

I'm trying to get V2Ray working so that:

a) I can acheive better WireGuard VPN speeds (to / from my LAN)
b) Better UDP speeds for things like streaming?

## Specifications

  - **OpenMPTCProuter version:** 0.58.5
  - **OpenMPTCProuter VPS version:** OpenMPTCProuter VPS 0.1026
  - **OpenMPTCProuter VPS provider:** OVHcloud
  - **OpenMPTCProuter platform:** RPI4
  - **Country:** UK
 
## Information

First off - **Thank you** for your time and effort getting this working. It's had a huge benefit for myself and many others... living in an area with poor internet speeds and without expert linux know-how. This was only dreamed about just a few years ago. I managed to get something up and running just using linux packages and scripts - But nothing on this level and it was time consuming!

I've been trying to get V2Ray working as the default proxy for a couple of weeks now. I've tried various things such as:

1. Removing all entries from the 'Port Forwards' section in the firewall - Just in case i'm having port conflicts
2. Changing the congestion algorythm (bb2, cubic)
3. Changing the TCP scheduler (BLEST, ECF, Round-Robin)
4. **A complete factory reset:**

I've tried re-flashing the SD card to stock factory and starting again - Thinking maybe something has gone wrong somewhere in my settings or something has become corrupted.... And strangely it does actually work for a short time... I then carefully start altering settings - Such as adding things to OMR bypass, adding a few port forwarding rules, adding DDNS, so that if OMR goes down my WAN services will still work, and sure enough, after a reboot I start facing the same V2Ray issues. I don't know what else to try.

It just says 'V2Ray not running'

[image]

It looks as though V2Ray is constantly restarting, and taking its time coming up:

[image]

I'm aware I should attach 'text files' - I'm not sure if that means linking to them, or pasting them in this post, so I apologies if I've not done this properly, but here is the **system log:**

<a href="https://portfolio.jb-net.co.uk/shared/v2ray/v2ray-not-running-system-log.txt" target="_blank">System Log Text File</a> *- opens in new tab*

And here is **'All Settings':**

<a href="https://portfolio.jb-net.co.uk/shared/v2ray/v2ray-not-running-all-settings.txt" target="_blank">All Settings Text File</a> *- opens in new tab*

Any help would be much appreciated, and again, thank you!

jamesbrindle, 2022-02-28:
Well... Don't all answer at once

Ysurac, 2022-02-28:
You can check ```/var/log/daemon.log on VPS```.
An iptables rule seems to not apply correctly, did you install any additional package ?
You can also try latest beta.

jamesbrindle, 2022-02-28:
Hey,

Thanks for that.

I have installed additional packages yes, these:

* GIT
* DDNS
* FTP Server
* Collectd - Statistics

I couldn't find anything particular in `/var/log/daemon.log` I'm afraid...

Also, I tried the latest beta. Everything was going OK and seemingly a lot quicker and smoother, but I then ran into the same issue when I began adding to the configs.

I have, however, managed to replicate the issue and I think I know vaguely what's causing it - It's the OMR-Bypass. Everything was going fine until I started to add rows into the OMR-Bypass config (via the Luci UI).

When I removed all items from OMR-Bypass, the problem went away.

It "seems" to be OK when adding full LAN IP addresses to ignore (my TV), but as soon as I start adding protocols and domains - the problem arises after a reboot.

This problem only seems to happen when the V2Ray proxy is used as the default.

Since OMR-Bypass is more important to me than the probable UDP performance gains from using V2Ray as opposed to shadowsocks, I've elected go back the 0.58.5 build and wait until a fix for this comes to fruition.

Thanks for your help, your software and your time.

Ysurac, 2022-03-03:
Can you give me the result of ```uci show omr-bypass`` via SSH on the router part ?

jamesbrindle, 2022-03-04:
Yep, no problem:

omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.amazonvideo=proto
omr-bypass.amazonvideo.url='cloudfront.net' 'llnw.net'
omr-bypass.lo=interface
omr-bypass.lo.id='5'
omr-bypass.eth0=interface
omr-bypass.eth0.id='6'
omr-bypass.wan1=interface
omr-bypass.wan1.id='3'
omr-bypass.wan2=interface
omr-bypass.wan2.id='4'
omr-bypass.wan3=interface
omr-bypass.wan3.id='8'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1200'
omr-bypass.wlan0=interface
omr-bypass.wlan0.id='11'
omr-bypass.WireGuardVPN=interface
omr-bypass.WireGuardVPN.id='8'
omr-bypass.WGVPN=interface
omr-bypass.WGVPN.id='10'
omr-bypass.@dpis[0]=dpis
omr-bypass.@dpis[0].proto='amazonvideo'
omr-bypass.@dpis[1]=dpis
omr-bypass.@dpis[1].proto='netflix'
omr-bypass.@dpis[2]=dpis
omr-bypass.@dpis[2].proto='disneyplus'
omr-bypass.@dpis[3]=dpis
omr-bypass.@dpis[3].proto='spotify'
omr-bypass.@ips[0]=ips
omr-bypass.@ips[0].ip='45.60.74.52'
omr-bypass.@ips[0].note='gov.uk'
omr-bypass.@ips[1]=ips
omr-bypass.@ips[1].ip='205.251.197.92'
omr-bypass.@ips[1].note='Netflix'
omr-bypass.@ips[2]=ips
omr-bypass.@ips[2].ip='205.251.192.81'
omr-bypass.@ips[2].note='Netflix'
omr-bypass.@ips[3]=ips
omr-bypass.@ips[3].note='Netflix'
omr-bypass.@ips[3].ip='205.251.194.147'
omr-bypass.@ips[4]=ips
omr-bypass.@ips[4].ip='205.251.199.192'
omr-bypass.@ips[4].note='Netflix'
omr-bypass.@ips[5]=ips
omr-bypass.@ips[5].ip='52.0.131.132'
omr-bypass.@ips[5].note='Netflix'
omr-bypass.@ips[6]=ips
omr-bypass.@ips[6].ip='3.221.228.214'
omr-bypass.@ips[6].note='Netflix'
omr-bypass.@ips[7]=ips
omr-bypass.@ips[7].ip='18.207.84.236'
omr-bypass.@ips[7].note='Netflix'
omr-bypass.@lan_ip[0]=lan_ip
omr-bypass.@lan_ip[0].ip='192.168.1.25'
omr-bypass.@lan_ip[0].note='GoogleTV'
omr-bypass.@macs[0]=macs
omr-bypass.@macs[0].mac='F8:4E:17:47:ED:D5'
[... 177 lines cut ...]
omr-bypass.@asns[5].note='Netflix'
omr-bypass.@asns[6]=asns
omr-bypass.@asns[6].asn='16509'
omr-bypass.@asns[6].note='Amazon'
omr-bypass.@asns[7]=asns
omr-bypass.@asns[7].asn='AS16509'
omr-bypass.@asns[7].note='Amazon'
omr-bypass.@asns[8]=asns
omr-bypass.@asns[8].asn='14618'
omr-bypass.@asns[8].note='HBOMAX'
omr-bypass.@asns[9]=asns
omr-bypass.@asns[9].asn='AS14618'
omr-bypass.@asns[9].note='HBOMAX'
omr-bypass.@asns[10]=asns
omr-bypass.@asns[10].asn='23286'
omr-bypass.@asns[10].note='Hulu'
omr-bypass.@asns[11]=asns
omr-bypass.@asns[11].asn='AS23286'
omr-bypass.@asns[11].note='Hulu'
omr-bypass.@asns[12]=asns
omr-bypass.@asns[12].asn='10753'
omr-bypass.@asns[12].note='National Lottery'
omr-bypass.@asns[13]=asns
omr-bypass.@asns[13].asn='AS10753'
omr-bypass.@asns[13].note='National Lottery'

jamesbrindle, 2022-03-04:
I've been trying to analyse the system log a little more...

There definitely seems to be a bit of a cat and mouse scenario of 'Starting OMR-Bypass', then stating V2Ray, then OMR-Bypass failing, so stating OMR-Bypass again, then V2Ray going down... and so on:

```
Fri Mar  4 03:09:54 2022 user.notice omr-bypass: OMR-ByPass is running
Fri Mar  4 03:09:55 2022 daemon.info v2ray[13036]: V2Ray 4.35.1 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.15.2 linux/arm64)
Fri Mar  4 03:09:55 2022 daemon.info v2ray[13036]: A unified platform for anti-censorship.
Fri Mar  4 03:09:55 2022 daemon.info v2ray[13036]: 2022/03/04 03:09:55 [Info] github.com/v2fly/v2ray-core/v4/main/jsonem: Reading config: /var/etc/v2ray/v2ray.main.json
Fri Mar  4 03:09:55 2022 user.notice v2ray: Rules DOWN
Fri Mar  4 03:09:55 2022 user.notice v2ray: Rules UP
Fri Mar  4 03:09:56 2022 user.notice v2ray: Reload omr-bypass rules
Fri Mar  4 03:09:56 2022 user.notice omr-bypass: Starting OMR-ByPass...
Fri Mar  4 03:10:00 2022 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 77.88.55.77)
Fri Mar  4 03:10:57 2022 daemon.info dnsmasq-dhcp[9116]: DHCPDISCOVER(eth0) 7c:a6:b0:0e:50:a8
Fri Mar  4 03:10:57 2022 daemon.info dnsmasq-dhcp[9116]: DHCPOFFER(eth0) 192.168.1.146 7c:a6:b0:0e:50:a8
Fri Mar  4 03:10:57 2022 daemon.info dnsmasq-dhcp[9116]: DHCPREQUEST(eth0) 192.168.1.146 7c:a6:b0:0e:50:a8
Fri Mar  4 03:10:57 2022 daemon.info dnsmasq-dhcp[9116]: DHCPACK(eth0) 192.168.1.146 7c:a6:b0:0e:50:a8
Fri Mar  4 03:11:11 2022 daemon.info dnsmasq-dhcp[9116]: DHCPREQUEST(eth0) 192.168.1.190 14:eb:b6:58:da:7c
Fri Mar  4 03:11:11 2022 daemon.info dnsmasq-dhcp[9116]: DHCPACK(eth0) 192.168.1.190 14:eb:b6:58:da:7c Archer_AX73
Fri Mar  4 03:11:16 2022 user.notice omr-bypass: Reload dnsmasq...
Fri Mar  4 03:11:16 2022 daemon.info dnsmasq[9116]: read /etc/hosts - 4 addresses
Fri Mar  4 03:11:16 2022 daemon.info dnsmasq[9116]: read /tmp/hosts/dhcp.cfg01411c - 25 addresses
Fri Mar  4 03:11:16 2022 daemon.info dnsmasq-dhcp[9116]: read /etc/ethers - 0 addresses
Fri Mar  4 03:11:16 2022 user.notice omr-bypass: OMR-ByPass is running
Fri Mar  4 03:11:16 2022 user.notice v2ray: Reload omr-bypass rules
Fri Mar  4 03:11:16 2022 user.notice omr-bypass: Starting OMR-ByPass...
Fri Mar  4 03:11:17 2022 user.notice post-tracking-post-tracking: Can't find omr-bypass rules, restart omr-bypass...

.
.
.

Fri Mar  4 03:12:39 2022 user.notice omr-bypass: Starting OMR-ByPass...
Fri Mar  4 03:12:39 2022 daemon.err omr-tracker[20333]: iptables-restore: line 2 failed
Fri Mar  4 03:12:40 2022 daemon.err omr-tracker[20333]: iptables-restore: line 2 failed
Fri Mar  4 03:12:40 2022 user.notice post-tracking-post-tracking: Can't find V2Ray, restart it...

.
.
.

Fri Mar  4 03:12:59 2022 user.notice omr-bypass: OMR-ByPass is running
Fri Mar  4 03:13:00 2022 daemon.info v2ray[4801]: V2Ray 4.35.1 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.15.2 linux/arm64)
Fri Mar  4 03:13:00 2022 daemon.info v2ray[4801]: A unified platform for anti-censorship.
Fri Mar  4 03:13:00 2022 daemon.info v2ray[4801]: 2022/03/04 03:13:00 [Info] github.com/v2fly/v2ray-core/v4/main/jsonem: Reading config: /var/etc/v2ray/v2ray.main.json
Fri Mar  4 03:13:00 2022 daemon.info v2ray: DNS disabled: main_dns
Fri Mar  4 03:13:00 2022 daemon.err omr-tracker[20335]: [info] DNS disabled: main_dns
Fri Mar  4 03:13:00 2022 daemon.info v2ray: Setting transparent proxy on port: 1897
Fri Mar  4 03:13:00 2022 daemon.err omr-tracker[20335]: [info] Setting transparent proxy on port: 1897
Fri Mar  4 03:13:00 2022 daemon.info v2ray: Transparent proxy mode: default
Fri Mar  4 03:13:00 2022 daemon.err omr-tracker[20335]: [info] Transparent proxy mode: default
Fri Mar  4 03:13:00 2022 user.notice v2ray: add rules
Fri Mar  4 03:13:00 2022 daemon.info dnsmasq-dhcp[3807]: DHCPREQUEST(eth0) 192.168.1.190 14:eb:b6:58:da:7c
Fri Mar  4 03:13:00 2022 daemon.info dnsmasq-dhcp[3807]: DHCPACK(eth0) 192.168.1.190 14:eb:b6:58:da:7c Archer_AX73
Fri Mar  4 03:13:01 2022 daemon.err omr-tracker[20335]: Flush terminated
Fri Mar  4 03:13:01 2022 daemon.err omr-tracker[20335]: ss-rules6: unknown option def
Fri Mar  4 03:13:01 2022 user.notice v2ray: Reload omr-bypass rules
Fri Mar  4 03:13:01 2022 user.notice omr-bypass: Starting OMR-ByPass...
Fri Mar  4 03:13:01 2022 daemon.info dnsmasq-dhcp[3807]: DHCPREQUEST(eth0) 192.168.1.190 14:eb:b6:58:da:7c
Fri Mar  4 03:13:01 2022 daemon.info dnsmasq-dhcp[3807]: DHCPACK(eth0) 192.168.1.190 14:eb:b6:58:da:7c Archer_AX73
Fri Mar  4 03:13:01 2022 daemon.info dnsmasq-dhcp[3807]: DHCPDISCOVER(eth0) 192.168.1.190 14:eb:b6:58:da:7c
Fri Mar  4 03:13:01 2022 daemon.info dnsmasq-dhcp[3807]: DHCPOFFER(eth0) 192.168.1.190 14:eb:b6:58:da:7c
Fri Mar  4 03:13:01 2022 daemon.info dnsmasq-dhcp[3807]: DHCPREQUEST(eth0) 192.168.1.190 14:eb:b6:58:da:7c
Fri Mar  4 03:13:01 2022 daemon.info dnsmasq-dhcp[3807]: DHCPACK(eth0) 192.168.1.190 14:eb:b6:58:da:7c Archer_AX73
Fri Mar  4 03:13:02 2022 daemon.err omr-tracker[20335]: iptables-restore: line 2 failed
Fri Mar  4 03:13:02 2022 daemon.err omr-tracker[20335]: iptables-restore: line 2 failed
Fri Mar  4 03:13:02 2022 daemon.err omr-tracker[20335]: iptables-restore: line 2 failed
Fri Mar  4 03:13:03 2022 daemon.err omr-tracker[20335]: iptables-restore: line 2 failed
Fri Mar  4 03:13:03 2022 daemon.err omr-tracker[20335]: iptables-restore: line 2 failed
Fri Mar  4 03:13:03 2022 daemon.err omr-tracker[20335]: iptables-restore: line 2 failed
[... cut ...]
Not too sure what `iptables-restore: line 2 failed` is referring to either and also... Something that's repeating: when V2Ray as proxy: DHCPACK:

```
DHCPACK(eth0) 192.168.1.190 14:eb:b6:58:da:7c Archer_AX73
```

This is the router (Archer_AX73) in 'AP' mode, using just to supply wifi... Oddly though, I've assigned this AP a static IP of: 192.168.1.70... and I can access it on that IP, yet, OpenMPTCProuter is also giving it (DHCP) an IP address of 192.168.1.190... Not entirely sure why and I can't access it on that IP either... Don't know if it's at all related?

Kind regards,

Jamie

closed 2022-06-07 by github-actions[bot]