Ysurac/openmptcprouter #2285: Raspberry Pi 4 gpio fan contol

skmshd, 2022-02-27:
I installed openmptcprouter to my raspberry pi4 and now want to turn on/off fan automatically.
But, it looks following kernel modules are not available in opkg list. Any ways to install them or enable gpio fan control ? 

- kmod-hwmon-core
- kmod-hwmon-gpiofan

xorbital, 2022-02-27:
I was using an actively cooled case for a few weeks and it worked flawlessly, did you install `kmod-hwmon-gpiofan` after **updating the package lists**? add the code to /boot/config.txt and rebooted?
`dtoverlay=gpio-fan,gpiopin=12,temp=55000`
Where `gpiopin` is the pin of your transistor switching the fan and `temp` is the temperature in °C * 1000

skmshd, 2022-02-28:
Thanks for your comment, I tried but I could not find kmod-hwmon-gpiofan.
Since "opkg update" command had some error, that's why?
When I use wget to check the errored packages, it returned "Invalid SSL" error.

```
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.58.5
 TARGET:      rpi4
 ARCH:        aarch64_cortex-a72

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Sat Jul 10 17:04:09 UTC 2021
-----------------------------------------------------
root@OpenMPTCProuter:~# opkg update
Downloading https://packages.openmptcprouter.com/v0.58.5/aarch64_cortex-a72/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://packages.openmptcprouter.com/v0.58.5/aarch64_cortex-a72/luci/Packages.sig
Signature check passed.
Downloading https://packages.openmptcprouter.com/v0.58.5/aarch64_cortex-a72/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://packages.openmptcprouter.com/v0.58.5/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading https://packages.openmptcprouter.com/v0.58.5/aarch64_cortex-a72/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://packages.openmptcprouter.com/v0.58.5/aarch64_cortex-a72/base/Packages.sig
Signature check passed.
Downloading https://packages.openmptcprouter.com/v0.58.5/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://packages.openmptcprouter.com/v0.58.5/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading https://packages.openmptcprouter.com/v0.58.5/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://packages.openmptcprouter.com/v0.58.5/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Downloading https://download.openmptcprouter.com/release/v0.58.5/rpi4/targets/bcm27xx/bcm2711/packages/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.58.5/rpi4/targets/bcm27xx/bcm2711/packages/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/base/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/base/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/luci/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/luci/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/openmptcprouter/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/openmptcprouter/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/packages/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/packages/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.58.5/rpi4/targets/bcm27xx/bcm2711/packages/Packages.gz, wget returned 5.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/base/Packages.gz, wget returned 5.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/luci/Packages.gz, wget returned 5.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/openmptcprouter/Packages.gz, wget returned 5.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.58.5/rpi4/packages/aarch64_cortex-a72/packages/Packages.gz, wget returned 5.
root@OpenMPTCProuter:~# opkg list |grep hwmon
root@OpenMPTCProuter:~# 
```

```
root@OpenMPTCProuter:~# wget https://download.openmptcprouter.com/release/v0.58.5/rpi4/targets/bcm27xx/bcm2711/packages/Packages.gz
Downloading 'https://download.openmptcprouter.com/release/v0.58.5/rpi4/targets/bcm27xx/bcm2711/packages/Packages.gz'
Connecting to 199.254.199.55:443
Connection error: Invalid SSL certificate
```

Ysurac, 2022-02-28:
Check this for the error: https://github.com/Ysurac/openmptcprouter/issues/2099

skmshd, 2022-03-05:
Thanks you so much for your help, Ysurac.
After resolving my opkg issue, I could install kmod-hwmon-gpiofan and auto fan control worked :)

closed 2022-03-05 by skmshd