Ysurac/openmptcprouter #2286: Banana Pi R64: How install OMR on SD Card or eMMC memory

ImprezyLive, 2022-02-27:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

I got Banan Pi R64, now i can run from tFTP initramfs.

**Describe why you are asking for this support?**

how install on SD card or eMMC memory?

**Additional context**

Ysurac, 2022-02-28:
You can follow OpenWrt doc: https://openwrt.org/toh/sinovoip/bananapi_bpi-r64_v1.1

ImprezyLive, 2022-02-28:
Ok, i install OpenWRT on eMMC, now can run OpenWRT, but after sysupgrade via openmptcprouter-v0.58.5-r0+16336-b36068d35d-mediatek-mt7622-bananapi_bpi-r64-squashfs-sysupgrade.itb nothing to change, still running OpenWRT.
I must some change ?

ImprezyLive, 2022-02-28:
I'm install openmptcprouter-v0.58.5-r0+16336-b36068d35d-mediatek-mt7622-bananapi_bpi-r64-sdcard.img.gz on SD card, runing i got after run boot:
tFtp 192.168.1.254: openwrt-mediatek-mt7622-bananapi_bpi-r64-initramfs-recovery.itb
Why download, why not run from SD Card ?

ImprezyLive, 2022-02-28:
Ok, i run initramfs and connect via COM interface.
What i got:
BusyBox v1.34.1 (2021-12-29 13:06:10 UTC) built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|
       |_|
------------------------------------------------------------------------------
   (r0+18442-02de391b08)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.58.5
 TARGET:      bpi-r64
 ARCH:        aarch64_cortex-a53

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Sun Feb 20 10:54:00 UTC 2022
-----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenMPTCProuter:~#
-ash: o: not found
root@OpenMPTCProuter:~# ls
-ash: l: not found
root@OpenMPTCProuter:~#
-ash: s: not found
root@OpenMPTCProuter:~#

I typing 'root', to login, next click on keyboard is Enter i got: '-ash: o: not found', typing 'ls', openwrt return '-ash: l: not found'.
Maybe some one can help select which image i must use to run?

ImprezyLive, 2022-02-28:
I found some information with file sdcardimage.img
After write this image to SD card and run, i can run openmptcprouter-v0.59beta3-5.15-r0+18442-02de391b08-mediatek-mt7622-bananapi_bpi-r64-initramfs-recovery.itb, system starts.
I can login via serial port, but if use 'original' sd card image is not possible to login via serial port.

ImprezyLive, 2022-03-01:
Compiler error
make[3]: *** [Makefile:235: /home/~/openmptcprouter/bpi-r64/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/iproute2-ipfull/iproute2-5.11.0/.built] Error 2

Kirball, 2022-04-21:
I have purchased a banana pi r64 as well, do I not just use the img file for the SD card to run OMR or do I also need to install OpenWRT.  I'm so confused as to what I need to do to get this working.  Has anyone got this working on a Bpi r64?

closed 2022-07-27 by github-actions[bot]