Ysurac/openmptcprouter #2291: RTL8211DN r8169 Kernel Panic

Anonymousey, 2022-03-06:
## Expected Behavior
No Kernel Panic when connected via shadowsocks 

## Current Behavior

Same issue as https://github.com/Ysurac/openmptcprouter/issues/588

expect kernel panic occurs on router side and not VPS side. used AMD and Intel hardware with same realtek card and same issue, suspected problem is the realtek driver/card?

## Possible Solution

## Steps to Reproduce the Problem

  1. 1 WAN Connected
  2. Run speedtest on client when router configured to use Shadowsocks
  3. Slow speed then kernel panic on router side

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.58.5 X64
  - OpenMPTCProuter VPS version: Unknown 
  - OpenMPTCProuter VPS provider: https://www.mlnl.host/
  - OpenMPTCProuter platform:  X86_64
  - Country:  UK, Vodafone Mobile

Ysurac, 2022-03-06:
I would need the log.

Anonymousey, 2022-03-06:
> I would need the log.

give a moment to get it for you

Anonymousey, 2022-03-06:
`[   12.760668] RTL8211DN Gigabit Ethernet r8169-100:00: attached PHY driver [RTL8211DN Gigabit Ethernet] (mii_bus:phy_addr=r8169-100:00, irq=IGNORE)
[   13.010311] r8169 0000:01:00.0 eth0: Link is Down
[   15.434420] r8169 0000:01:00.0 eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   50.520188] ------------[ cut here ]------------
[   50.520990] WARNING: CPU: 0 PID: 20412 at skb_try_coalesce+0x3f4/0x450
[   50.521800] Modules linked in: ath9k ath9k_htc ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial pppoe ppp_async option iwlmvm iwldvm ipw cdc_mbim ath9k_hw ath wireguard usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 smsc95xx smsc75xx sierra_net sierra rtlwifi rtl8xxxu rtl8812au rtl8187 rt2x00usb rt2x00lib rndis_host qmi_wwan pppox ppp_generic pl2303 oti6858 mt7601u mos7720 mct_u232 mcs7830 mac80211 lzo libchacha20poly1305 libblake2s keyspan kalmia iwlwifi ipt_REJECT huawei_cdc_ncm garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cypress_m8 curve25519_x86_64 cp210x chacha_x86_64 ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem blake2s_x86_64 belkin_sa ax88179_178a asix ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark
[   50.521844]  xt_connlimit xt_connbytes xt_condition xt_comment xt_cgroup xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor via_velocity via_rhine usbserial usbnet tulip ts_fsm ts_bm solos_pci slhc sky2 skge sis900 sis190 sch_cake rtl8150 r8712u(C) r8169 r8152 r8125 r6040 poly1305_x86_64 pegasus pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci mmc_spi macvlan lzo_decompress lzo_compress libcurve25519_generic libchacha libblake2s_generic kaweth iptable_raw iptable_nat iptable_mangle
[   50.521996]  iptable_filter ipt_ah ipt_ECN ipheth ip6table_raw ip_tables hso forcedeth ezusb ethoc et131x ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 e1000e e100 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_acm bnx2 atlantic atl2 atl1e atl1c atl1 asn1_decoder arptable_filter arpt_mangle arp_tables aquantia 8390 8139too 8139cp tcp_nanqinlang tcp_bbr2 sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred configs evdev i2c_gpio i2c_dev spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark
[   50.521996]  ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig bonding 3c59x ip6_gre ip_gre gre ixgbevf ixgbe igb e1000 ifb nat46 ip6_udp_tunnel udp_tunnel sit sctp libcrc32c mdio ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel netlink_diag tunnel6 tunnel4 ip_tunnel hfcpci hfcmulti veth tun mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key xfrm_algo autofs4 br2684 atm nls_utf8 zram zsmalloc natsemi eeprom_93cx6 ena sha256_ssse3 sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg md5 kpp hmac ghash_generic gcm echainiv des_generic libdes deflate ctr cbc authenc crypto_acompress nls_iso8859_1 nls_cp437 vfat fat sdhci_pltfm xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl gpio_button_hotplug button_hotplug tg3 b44 ssb ptp realtek broadcom
[   50.521996]  bcm_phy_lib pps_core mii libphy
[   50.521996] CPU: 0 PID: 20412 Comm: sh Tainted: G         CI       5.4.106 #0
[   50.521996] Hardware name: LENOVO 1200CTO/1200CTO, BIOS 8IET34WW(1.07) 07/28/2011
[   50.521996] RIP: 0010:skb_try_coalesce+0x3f4/0x450
[   50.521996] Code: 0e f8 48 89 7c 0a f8 48 8d 7a 08 48 83 e7 f8 48 29 fa 01 d0 48 29 d6 c1 e8 03 89 c1 f3 48 a5 45 0f b6 6c 24 02 e9 06 ff ff ff <0f> 0b e9 ce fe ff ff 48 8d 46 ff 48 8b 76 ff 48 c1 ee 2d 41 89 f5
[   50.521996] RSP: 0000:ffffc900000038d8 EFLAGS: 00010297
[   50.521996] RAX: 00000000000007d4 RBX: 0000000000000508 RCX: ffff888117a27200
[   50.521996] RDX: ffffc900000039c0 RSI: 00000000000000c0 RDI: 0000000000000000
[   50.521996] RBP: ffffc90000003908 R08: ffff888119c29d00 R09: ffff888119c29200
[   50.521996] R10: ffffc9000000391c R11: ffff8881160cfec0 R12: ffff888117a272c0
[   50.521996] R13: 0000000000000001 R14: 0000000000000508 R15: 00000000000004d4
[   50.521996] FS:  00007f64c6544d48(0000) GS:ffff88811c400000(0000) knlGS:0000000000000000
[... cut ...]
[   50.521996] RSP: 002b:00007ffd6dab0fc8 EFLAGS: 00000246 ORIG_RAX: ffffffffffffffd8
[   50.521996] RAX: 000000000000004e RBX: 00007f64c65456c8 RCX: 0000000000000000
[   50.521996] RDX: 00007f64c65455a0 RSI: 0000000002015c73 RDI: 00007f64c64d0920
[   50.521996] RBP: 00007f64c64d08e0 R08: 3d3d3d3d3d3d3d3d R09: 8080808080808080
[   50.521996] R10: 000000000000006e R11: 0000000000000002 R12: 0000000002015c73
[   50.521996] R13: 0000000000000100 R14: 00007f64c6545360 R15: 0000000002015c72
[   50.521996] ---[ end trace 268f84cd2283ebce ]---`

Anonymousey, 2022-03-06:
system log

Sun Mar  6 20:16:35 2022 daemon.notice netifd: Interface 'wan2' has link connectivity
Sun Mar  6 20:16:35 2022 kern.info kernel: [   15.434420] r8169 0000:01:00.0 eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=tty28, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty28
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tty28, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyS15, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS15
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS15, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=tty56, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty56
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tty56, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=tty18, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty18
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tty18, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=tty1, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty1
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tty1, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=tty46, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty46
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tty46, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice mptcp: Reloading mptcp config due to ifup of wan1 (wan1)
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyS6, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS6
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS6, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=tty36, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty36
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tty36, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=tty26, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty26
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tty26, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyS13, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS13
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS13, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=tty54, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty54
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tty54, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=tty16, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty16
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tty16, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=tty44, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty44
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: event reported: action=add, name=tty44, subsystem=tty
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  6 20:16:36 2022 user.notice ModemManager: hotplug: cached event found: action=add, name=console, subsystem=tty, sysfspath=/sys/devices/virtual/tty/console
[... cut ...]
Sun Mar  6 20:18:17 2022 user.notice post-tracking-post-tracking: Set firewall on server
Sun Mar  6 20:18:17 2022 user.notice omr-bypass: Starting OMR-ByPass...
Sun Mar  6 20:18:18 2022 user.notice omr-bypass: Reload dnsmasq...
Sun Mar  6 20:18:19 2022 daemon.info dnsmasq[22091]: read /etc/hosts - 4 addresses
Sun Mar  6 20:18:19 2022 daemon.info dnsmasq[22091]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Mar  6 20:18:19 2022 daemon.info dnsmasq-dhcp[22091]: read /etc/ethers - 0 addresses
Sun Mar  6 20:18:19 2022 user.notice omr-bypass: OMR-ByPass is running

Anonymousey, 2022-03-06:
I found this if this helps as well. when searching google for log terms

https://github.com/openwrt/mt76/issues/121

closed 2022-06-11 by github-actions[bot]