Ysurac/openmptcprouter #2293: Pi-hole installation script not accessible due to certificate error

thomasgx, 2022-03-13:
## Expected Behavior
Installation of pi hole

## Current Behavior

Can't go through pi-hole sh script :

`root@vps:~# wget -O - https://www.openmptcprouter.com/server/omr-pihole.sh | sh
--2022-03-13 08:26:44--  https://www.openmptcprouter.com/server/omr-pihole.sh
Résolution de www.openmptcprouter.com (www.openmptcprouter.com)… 2001:ac8:25:da::2, 89.44.9.58
Connexion à www.openmptcprouter.com (www.openmptcprouter.com)|2001:ac8:25:da::2|:443… connecté.
Erreur : le certificat de « www.openmptcprouter.com » n’est pas de confiance.
Erreur : le certificat de « www.openmptcprouter.com » a expiré.
`

## Possible Solution

Problem With the certificate ?

## Steps to Reproduce the Problem

  1. Ssh to VPs
2. Launch pi hole script
3. Error

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  0.57.3
  - OpenMPTCProuter VPS version:  0.1024
  - OpenMPTCProuter VPS provider: firsheberg
  - OpenMPTCProuter platform:  RPI3
  - Country:  FR

Ysurac, 2022-03-13:
Certificates are generated by the CDN, possible that a node wasn't updated. Try again.
Else check VPS date.

brackerp, 2022-05-15:
tried 2 weeks on sunday a week apart got same error

Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|68.70.205.2|:443... connected.
ERROR: The certificate of ‘www.openmptcprouter.com’ is not trusted.
ERROR: The certificate of ‘www.openmptcprouter.com’ has expired.

Ysurac, 2022-05-15:
Check date time of your server.

brackerp, 2022-05-15:
other then the router at my house are at diffrent time zones the time is the same
[image]

Ysurac, 2022-05-15:
Check that VPS ca certificates are updated.
You can use "--no-check-certificate" if you are sure there is no middle box that can make/monitor/change the certificate to invalid.
I don't manage "www.openmptcprouter.com" certificates, the CDN (keycdn) is managing them.

brackerp, 2022-05-15:
it looks like pihole trys to update also

root@localhost:~# wget --no-check-certificate -O - http://www.openmptcprouter.com/server/omr-pihole.sh | sh
--2022-05-15 12:20:09--  http://www.openmptcprouter.com/server/omr-pihole.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 68.70.205.2, 68.70.205.1, 68.70.205.4, ...
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|68.70.205.2|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://www.openmptcprouter.com/server/omr-pihole.sh [following]
--2022-05-15 12:20:09--  https://www.openmptcprouter.com/server/omr-pihole.sh
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|68.70.205.2|:443... connected.
WARNING: The certificate of ‘www.openmptcprouter.com’ is not trusted.
WARNING: The certificate of ‘www.openmptcprouter.com’ has expired.
HTTP request sent, awaiting response... 200 OK
Length: 2584 (2.5K) [application/octet-stream]
Saving to: ‘STDOUT’

-                                   100%[=================================================================>]   2.52K  --.-KB/s    in 0s

2022-05-15 12:20:09 (66.2 MB/s) - written to stdout [2584/2584]

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
You can select any interface and set any IPs during Pi-hole configuration, this will be modified for OpenMPTCProuter at the end.
Don't apply Pi-hole firewall rules.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Run Pi-hole install script...

  [✓] Root user check

        .;;,.
        .ccccc:,.
         :cccclll:.      ..,,
          :ccccclll.   ;ooodc
           'ccll:;ll .oooodc
             .;cll.;;looo:.
                 .. ','.
                .',,,,,,'.
              .',,,,,,,,,,.
            .',,,,,,,,,,,,....
          ....''',,,,,,,'.......
        .........  ....  .........
        ..........      ..........
        ..........      ..........
        .........  ....  .........
          ........,,,,,,,'......
            ....',,,,,,,,,,,,.
               .',,,,,,,,,'.
                .',,,,,,'.
                  ..'''.

  [✗] Update local cache of available packages
  Error: Unable to update package cache. Please try "sudo apt update"
Done
-------------------------------------------------------------------------------------------------------------------------------
OMR Pi-hole configuration...
sh: 27: cannot create /etc/lighttpd/external.conf: Directory nonexistent
Failed to restart lighttpd.service: Unit lighttpd.service not found.
sh: 35: cannot create /etc/pihole/setupVars.new.conf: Directory nonexistent
mv: cannot stat '/etc/pihole/setupVars.new.conf': No such file or directory
sh: 37: cannot create /etc/pihole/setupVars.conf: Directory nonexistent
sh: 43: cannot create /etc/dnsmasq.d/01-pihole.new.conf: Directory nonexistent
mv: cannot stat '/etc/dnsmasq.d/01-pihole.new.conf': No such file or directory
sh: 45: cannot create /etc/dnsmasq.d/99-omr.conf: Directory nonexistent
Failed to restart pihole-FTL.service: Unit pihole-FTL.service not found.
Done
======================================================================================================================================
To use Pi-hole in OpenMPTCProuter, you need to 'Save & Apply' the wizard again in System->OpenMPTCProuter then reboot OpenMPTCProuter.
Web interface will be available on 10.255.255.1 if you use Glorytun TCP, 10.255.254.1 if you use Glorytun UDP.
======================================================================================================================================
root@localhost:~#

is there a email to let the CDN know there is a issue

brackerp, 2022-05-15:
root@localhost:~# apt update
Hit:1 http://security.debian.org/debian-security buster/updates InRelease
Hit:2 http://deb.debian.org/debian buster-backports InRelease
Ign:3 https://repo.openmptcprouter.com buster InRelease
Hit:4 http://ftp.us.debian.org/debian buster InRelease
Err:5 https://repo.openmptcprouter.com buster Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 68.70.205.3 443]
Hit:6 http://ftp.us.debian.org/debian buster-updates InRelease
Hit:7 http://httpredir.debian.org/debian stable InRelease
Reading package lists... Done
E: The repository 'https://repo.openmptcprouter.com buster Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
r

Ysurac, 2022-05-15:
What is the VPS provider ?

brackerp, 2022-05-15:
https://buyvm.net/kvm-dedicated-server-slices/

closed 2022-08-19 by github-actions[bot]