Ysurac/openmptcprouter #2296: Running OMR on a Freebox Delta - Tutorial

nbald, 2022-03-16:
# Introduction

I was able to run OMR on the Freebox Delta, using the native VM capabilities.

This is interesting because we do not need additional hardware. The Freebox Delta has a quite powerful CPU (4-cores ARMv8) and good network connectivity (4x 1G + 1x 10G SFP+) .

My first impression is that OMR works flawlessy on that platform.

[image]

# Image compilation

Those steps are to be run on your computer.
We assume you are using Ubuntu 20.04. It might work with other distributions.

## Prerequisites

```
sudo apt-get install busybox curl rsync build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler wget tar
```

## Getting OMR sources

```
BRANCH=master # can also be a tagged branch like "v0.58.5"
git clone https://github.com/Ysurac/openmptcprouter.git
cd openmptcprouter
git checkout $BRANCH
OMR_TARGET="fbxdelta" OMR_FEED_SRC="$BRANCH" ./build.sh
```
## Configuring the build

```
cd fbxdelta/source
```

We need to enable UEFI stub in the kernel, to be able to boot the VM.
Open file `target/linux/armvirt/config-5.4`, and append the following lines to the end of file:
```
CONFIG_EFI_STUB=y
CONFIG_EFI=y
CONFIG_EFI_VARS=y
CONFIG_ARCH_SUPPORTS_ACPI=y
CONFIG_ACPI=y
```

Then we need to configure the build
```
make menuconfig
```
[image]

Select the appropriate options for our platform:
* Target System: `QEMU ARM Virtual Machine`
* Subtarget: `QEMU ARMv8 Virtual Machine (cortex-a53)`. **NOT QEMU ARM Virtual Machine (cortex-a15)**

There is no other changes needed. Save the file to `.config` and exit the configuration menu.

This is it. You can now launch the build and take your afternoon to go skiing. The compilation takes some time.

```
make -j6
```

nbald, 2022-03-16:
# next steps

coming tomorrow

Ysurac, 2022-03-17:
I don't have a Freebox Delta, but if it's working correcly, I can create the dedicated config file and compile it automatically for each snapshot and releases.

jjak0b, 2022-04-15:
> I don't have a Freebox Delta, but if it's working correcly, I can create the dedicated config file and compile it automatically for each snapshot and releases.

It would be very useful adding images for armvirt. I compiled it last month to use as qemu vm on the raspberry pi and run very well with kvm enabled

vinceducat, 2022-04-19:
@Ysurac Freebox VM could be a good idea

nbald, 2022-05-06:
@ajoffre the second part of the tutorial is missing... 

Here are the important steps:

## create a 5 GB qcow2 disk (or smaller size):
`qemu-img create -f qcow2 mydisk.qcow2 5G`

## make the disk available for the system
```
sudo modprobe -av nbd
sudo qemu-nbd -c /dev/nbd0 ~/mydisk.qcow2
```

## partition the disk
`sudo fdisk /dev/nbd0`
- create a new GPT table
- create a first partition. 10MB, type = EFI
- create a second partition with all the remaining space, type = Linux filesystem
- write and quit
`

## copy the omr image to the second partition
```
gunzip openmptcprouter-v*-armvirt-64-rootfs-ext4.img.gz
dd if=openmptcprouter-v*-armvirt-64-rootfs-ext4.img of=/dev/nbd0p2
```

## prepare the boot partition
```
sudo mkfs.fat /dev/nbd0p1
sudo mkdir /mnt/omr-boot
sudo mount /dev/nbd0p1 /mnt/omr-boot
sudo echo "root=/dev/sda2 ro initrd=\initramfs-linux.img" > /mnt/omr-boot/options.txt
sudo umount /mnt/omr-boot
sudo rm -rf /mnt/omr-boot
```

## the qcow2 image is now ready. Unload the disk
`sudo qemu-nbd --disconnect /dev/nbd0`

## set-up the EFI boot

- load the qcow2 image into the freebox and boot the VM. I recommend to use 2 cpu cores.
- the UEFI boot will fail. You need to enter the UEFI shell and do the following:
- delete the PXE boot option
```
bcfg boot dump
  Option: 04 <--- we need that number
  UEFI PXEv4
bcfg boot rm 4 <--- the number we just got
```
- add a new boot option
```
bcfg boot add 0 FS0:\image "OMR"
bcfg boot -opt 0 FS0:\options.txt
```
(if FS0 does not work, try FS1)

If I did not forget anything, your VM should now be ready.

ajoffre, 2022-05-07:
Thank you very much Nicholas,
But I can't find the FI image (bcfg boot add 0 FS0:\image "OMR") that you are using.
On FS0 I only have the option.txt file.
Maybe I missed a step?

nbald, 2022-05-07:
I forgot the `sudo cp openmptcprouter-v*-armvirt-64-Image /mnt/omr-boot/image` in the boot partition's step.
I just edited the instructions. Note that you will need a bigger boot partition. I originally specified 10M, but it will not be enough. 100M is fine.

closed 2022-08-10 by github-actions[bot]

lewurm, 2022-09-09:
Thanks the instructions @nbald.  I wanted to create a build to run on `qemu-system-aarch64` with `edk2-armvirt` firmware images. When adding the boot entry in the UEFI shell I stumbled over those problems.

#### 1. Buggy `bcfg boot -opt` parsing

```
Shell> bcfg boot -opt 0 FS0:\options.txt
bcfg: Invalid argument - 'Option Index'
```
[This is a known bug](https://bugzilla.tianocore.org/show_bug.cgi?id=3080). As a workaround, the index can be provided in hexadecimal.
```
Shell> bcfg boot -opt 0x0 FS0:\options.txt
```

#### 2. `options.txt` must be in UTF-16LE

If you get something like this
> [    0.000000] Kernel command line: 潲瑯⼽敤⽶摶㉡爠⁯湩瑩摲尽湩瑩慲晭⵳楬畮⹸浩੧

and also OpenMPTCP not coming up due to missing the rootfs you probably haven't stored the `options.txt` as UTF-16 file. Best you check before packing up the image/boot partion:

```shell
$ iconv -f UTF-8 -t UTF-16 /mnt/omr-boot/options.txt > new.txt
$ sudo mv new.txt /mnt/omr-boot/options.txt
$ xxd /mnt/omr-boot/options.txt
00000000: fffe 7200 6f00 6f00 7400 3d00 2f00 6400  ..r.o.o.t.=./.d.
00000010: 6500 7600 2f00 7600 6400 6100 3200 2000  e.v./.v.d.a.2. .
00000020: 7200 6f00 2000 6900 6e00 6900 7400 7200  r.o. .i.n.i.t.r.
00000030: 6400 3d00 5c00 6900 6e00 6900 7400 7200  d.=.\.i.n.i.t.r.
00000040: 6100 6d00 6600 7300 2d00 6c00 6900 6e00  a.m.f.s.-.l.i.n.
00000050: 7500 7800 2e00 6900 6d00 6700 0a00       u.x...i.m.g...
```
Note the BOM `0xfffe` in the beginning of the file. Also, [adding two spaces in the beginning of the file](https://wiki.archlinux.org/title/EFISTUB#bcfg) helps to avoid confusion around the encoding. You can do this in the UEFI shell:
```
Shell> edit FS0:\options.txt
```
This editor shows you the current encoding in the top status bar (it should say `UNICODE` not `ASCII`). You can save and quit the editor with CTRL+Q. This is how it looked like for me in the end:

```
Shell> bcfg boot add 0x0 FS0:\image "OMR"
Target = 0004.
bcfg: Add Boot0004 as 0
Shell> bcfg boot -opt 0x0 FS0:\options.txt
Shell> bcfg boot dump -v
Option: 00. Variable: Boot0004   
  Desc    - OMR
  DevPath - PciRoot(0x0)/Pci(0x1,0x2)/Pci(0x0,0x0)/HD(1,GPT,B185E38A-AACD-5847-AB72-2B59B0F2F59F,0x800,0x32000)/\image
  Optional- Y
  00000000: FF FE 20 00 20 00 72 00-6F 00 6F 00 74 00 3D 00  *.. . .r.o.o.t.=.*
  00000010: 2F 00 64 00 65 00 76 00-2F 00 76 00 64 00 61 00  */.d.e.v./.v.d.a.*
  00000020: 32 00 20 00 72 00 6F 00-20 00 69 00 6E 00 69 00  *2. .r.o. .i.n.i.*
  00000030: 74 00 72 00 64 00 3D 00-5C 00 69 00 6E 00 69 00  *t.r.d.=.\.i.n.i.*
  00000040: 74 00 72 00 61 00 6D 00-66 00 73 00 2D 00 6C 00  *t.r.a.m.f.s.-.l.*
  00000050: 69 00 6E 00 75 00 78 00-2E 00 69 00 6D 00 67 00  *i.n.u.x...i.m.g.*
  00000060: 0A 00                                            *..*

```

pbonnard, 2024-10-05:
Hello, anyone has a pre-compiled image for Freebox/armvirt-64 to share?