Ysurac/openmptcprouter #2299: 1Gbps or More Aggregation

ccmks, 2022-03-18:
Has anyone ever achieved 1Gbps or more (assuming using 10Gbps VPS) aggregated speed on OMR? If so, I am curious about the minimum hardware requirement such as:

CPU
RAM
NIC

Thank you

Adorfer, 2022-03-20:
1GBit is easy, 2GBit/s feasable, 10GBit/s: no idea. 

Take a look at https://github.com/Ysurac/openmptcprouter/issues/1108

ccmks, 2022-03-20:
For 10Gbps, I am assuming only 5Gbps is being used effectively, as 10Gbps speed usually half-duplex connection...

QatarMo, 2022-04-05:
[image]

10g vps

router
10900K
16gb
Intel X520-DA2
2x 1G down 512 up Wan's

ccmks, 2022-04-12:
Is it 3X 1Gbps internet? Why is it not even close to 2.5Gbps?

How is the CPU load during the test?

closed 2022-07-17 by github-actions[bot]

Llorx, 2023-02-01:
> > Is it 3X 1Gbps internet? Why is it not even close to 2.5Gbps?
> > How is the CPU load during the test?
> 
> yes they are 1Gbps each,,load is like 30% on max on both router and vps i do get over 2G sometimes but i guess its the latency,its rock stable been running this for over a year now [image]

Just to clarify. Individual tests give 100 ping and 1gb but a mptcp test give also 100 ping and only 2gb?

I'm asking because I thought that the loss in bandwidth was way less. Losing 33% of the bandwidth is a real bummer. Have you tried other configurations just in case?

ccmks, 2023-02-02:
I haven't reach 1Gbps yet on aggregation, but so far only 180 Mbps and 300
Mbps I have seen being aggregated.

You also might need to check the connection between your ISP to your VPS to
see if you are getting full 1 Gbps.

Until now, I still don't know how to test for each individual ISP

Ysurac, 2023-02-02:
ping is a problem to get a good speed. If you have a good VPS, connections with low latency and a router with a good CPU you can reach 1Gbps without problem.
If you mean testing speed between VPS and router, there is 2 way : ```omr-test-speed-server``` or ```omr-iperf``` that use iperf server installed on VPS side.

Llorx, 2023-02-02:
@Ysurac so if you have a good CPU, ping variations shouldn't affect that much?

The problem I'm facing is saturated 4g. We stream sports events and some times there are more people than the 4g near stations can handle, so the ping and bandwidth fluctuates a lot. I want to use openmptcp with 4g, 5g and Starlink so when someone of them catches up, it works, but if I have 10mbps upload from Starlink and a fluctuating 4g, I don't want it to drop to 6mbps.

Ysurac, 2023-02-02:
In all case ping variation affect speed.
SQM with SQM autorate should help for connection that fluctuates.

Llorx, 2023-02-02:
Hm hmmmm. Ok, perfect! When I receive my router I'm going to do all the testing. Thank you!

QatarMo, 2023-02-02:
> Just to clarify. Individual tests give 100 ping and 1gb but a mptcp test give also 100 ping and only 2gb?
>
> I'm asking because I thought that the loss in bandwidth was way less. Losing 33% of the [...]

In my case Individual tests in a local test server have a ping of 2ms is 1gb/512 down/up speed and to the vps 100 is I have tested a lot and I think my bandwidth lose is due to ping/distance, no company lese vps with 10g nic locally so I had to look for the best ping vps

Llorx, 2023-02-03:
Ohhh ok ok, I understand it now. Perfect, thank you for your reply @QatarMo !