Ysurac/openmptcprouter #2300: OMR Max WAN

ccmks, 2022-03-19:
How many maximum WAN NIC can be supported current OMR, especially for X86_64?

Thank you

Ysurac, 2022-03-19:
on x86 and x86_64, 16 wans are supported, 8 on others platform.

ccmks, 2022-03-20:
I installed on Hyper-V with Gen2 selected and added 10 WAN. How come it can only see 8 WAN?

I am using v0.58.5. Any idea?

Ysurac, 2022-03-20:
Should be 16 wans on current 5.4 kernel.
What do you mean by "see" ? How do you check that ?

ccmks, 2022-03-20:
I went to SSH to OMR to check the number of interface read and only see up to eth8

Ysurac, 2022-03-20:
I don't have such interface limit, the 8 or 16 limits is about aggregated interfaces. Else you can have as many interfaces as you want.
If you don't see all physical interfaces via ```ip a``` this can be related to a missing driver.

closed 2022-06-24 by github-actions[bot]