Ysurac/openmptcprouter #2305: WAN Priority over other

ccmks, 2022-03-23:
Hello, I have scenario of 4 WAN where I would like the OMR to prioritize WAN 1 when do aggregation over the other. The reason that because WAN 1 has the largest data cap compare to other WAN.

In current situation, when aggregated, it is tend to use WAN 2 connection much more than other WAN. Is there anyway to configure some weight or manipulation that OMR will always using WAN1 when do aggregation?

I am currently running on v0.58.5

ccmks, 2022-03-23:
[image]

Please see this picture.

Thank you

Ysurac, 2022-03-23:
WAN are used based on latency, WAN with lowest latency is used first.
There is an option in Network->Interface to add some latency on a WAN, but it's more for test than daily use.

ccmks, 2022-03-23:
Upon checking, if lowest latency is being preferred, WAN3 should be the one being used most.

[image]

But in this case, it is using WAN2

ccmks, 2022-03-23:
I tried to increase WAN2 additional latency to 30, so should it be showing 60ms on OMR status page?

Ysurac, 2022-03-23:
1 ms is not a big difference, this can change all the times.
For wan3 and wan4, why does they have same ip and wan address ? provider do some CGNAT ?

Yes this should increase latency in status page (but I've not tested this since a long time...)

ccmks, 2022-03-23:
For WAN3 and WAN4 is 2 separate line of ISP that happened to share public IP.

I have increased to 30ms on WAN2 but on status page it doesn't increase the value to 60ms, instead it is still at 30ms

ccmks, 2022-03-23:
Besides manipulating the latency, is there any way to configure priority or weight on interface?

Ysurac, 2022-03-23:
no, for now only latency is used for priority.

ccmks, 2022-03-23:
Is there anyway to add request feature for this? This will be very helpful since we can decide which WAN we can manipulate, especially if the WAN has lower latency, but cost more to run due to metered data cap, and prefer other WAN that has higher latency but has more data cap.

ccmks, 2022-03-23:
I do noticed that if I add latency, the latency will be increased once I reboot OMR. Is there anyway to take effect without rebooting OMR? Also, I tried to restart the interface still not increasing latency.

Thank you

Ysurac, 2022-03-23:
I don't remember if I fixed this in develop branch, I will check.
Support for setting priority will not be in next release but it's on todo list, I think on 5.15 based release.

ccmks, 2022-03-23:
Appreciate your support!

closed 2022-07-11 by github-actions[bot]