Ysurac/openmptcprouter #2306: RasberryPi 4B - only 1Mb / second vs 10Mb unbonded - is this normal?

cannycookie, 2022-03-25:
Am i missing something?  Is this normal?

VPS Download Direct = 132MB/s
Virgin Media Direct Download = 13.3MB/s
OMR download (only with single WAN via virgin)  = 895KB/s

I expected to see some performance drop but 10x slower  seems a bit wrong. Anyone have any suggestions?

## Specifications
  - OpenMPTCProuter version: v0.58.5 r0
  - OpenMPTCProuter VPS version: 0.1026
  - OpenMPTCProuter VPS provider: AWS t3a small
  - OpenMPTCProuter platform: RPI 4B 8Gb

Ysurac, 2022-03-25:
Via SSH on the router, use ```omr-test-speed``` to test aggregated speed and ```omr-test-speed wan1``` to test wan1 speed directly. If this last test give a correct result, you can try to change TCP congestion control in Network->MPTCP.
Check also load average on VPS when you download.

cannycookie, 2022-03-25:
Still the same:-
  On virgin directly
```

  Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  6 9536M    6  587M    0     0  11.7M      0  0:13:34  0:00:50  0:12:44 11.9M
```

On Router Bonded & WAN connection - Cubic Congestion Control
```
Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  9 9536M    9  901M    0     0   895k      0  3:01:50  0:17:11  2:44:39  851k
```
Who doesn't love reading up on congestion control methods on a Friday night? I never knew there were so many techniques. 😅😅 

I've tried BBR, Balia, MCTCPDSYNC and there's no difference.

Two observations.  if i tracert directly to the target test server via virgin modem, it's just a few hops
[image]

But running via the VPS,  is much more complicated with many slow hops and there's a different EC2 instance listed each time, then a bunch of 100.xx steps.
[image]

I'm at the edge of my understanding now but a logical question...Is it possible that  using just 1 modem results in a bigger decrease in performance than if there were more?

Also ran HTOP on both servers whilst downloading and there was barely any change in utilisation, all low.

[image]

cannycookie, 2022-03-25:
PS.  Thank you so much @Ysurac for all your work, this is super cool.  If I get this working will buy you not 1 beer but many beers 😀

Ysurac, 2022-03-25:
All is green in status page ?
You can try to disable encryption in the wizard, "Advanced settings" checkbox.

cannycookie, 2022-03-25:
Thinking to try another VPS with a different provider as a first step towards identifying the bottleneck.

Next maybe to switch out the pi for something else or is it more likely to be a s/w configuration issue somewhere?

cannycookie, 2022-03-26:
Unofficial test results with no encryption from 3 different VPS providers

- AWS Small Instance 2 Cores 2Gb: **880K/s**
- Google Cloud Small 2 Cores 2Gb: **750K/s**
- OVH Cloud 2 Cores 4Gb: **915K/s**

Google box seemed slower during installation from the start, lots of installed google cloud extras - took ages to update.  No fan of google cloud interface either.
OVH has simpler GUI and definitely snappier box.  

Still no luck with finding the missing 9Mb though. Wondering if it's worth switching out the pi for a Virtual Box on my Mac, just to see.

cannycookie, 2022-03-26:
HAHAHA, problem solved, gotta have so much patience when learning new shit, been on this for 2 days solid, maybe 4 hours sleep.

I was using a second nic on a USB port for the modem.  Thought I'd just try it without and see what happens... 💥

```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 91 9536M   91 8755M    0     0  12.9M      0  0:12:15  0:11:15  0:01:00 13.0M
```

closed 2022-03-26 by cannycookie