Ysurac/openmptcprouter #2308: Source IP preservation through proxy

markfoodyburton, 2022-03-30:
I have an (e.g. web) server behind behind the mptcp router. However, the 'source' address that the (web) server sees is always the address of the mptcp router.

(Net result, things break - e.g. fail2ban isn't happy, etc etc....)

I believe this is because of the nature of the proxies - but I'm optimistic that there is a solution?

Cheers
Mark.

Ysurac, 2022-03-30:
I would add a reverse proxy on the VPS like nginx.

markfoodyburton, 2022-03-30:
Thanks @Ysurac - for http(s), that could work, I'm not sure it's a 'general' solution for other services though?
I guess I was optimistically hoping that openmptcprouter would solve the issue by 'magic'. :-)

markfoodyburton, 2022-03-31:
There's a simple fix for this - at least, it seems to work - simply dont add the SNAT in the howto :
https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

closed 2022-03-31 by markfoodyburton