Ysurac/openmptcprouter #2311: doing nat on omr are not copyed to vps

BillyMyIT, 2022-04-03:
## Expected Behavior
I put a nat inside the webinterface of omr to open a port to a device inside my lan
I expect that the vps will have the rule set automatically (I thought it's the default behavior but maybe i'm wrong)

## Current Behavior
I put my nat rule on omr
try to do a sudo nano /etc/shorewall/rule but the rule is not there
no info in system log

## Specifications

openmptcprouter v0.58.5 r0+16336-b36068d35d / LuCI Master git-21.077.59320-d1bf56d
  - OpenMPTCProuter VPS version: i don't remember how to have the version, uname -a give : Linux mwc383 5.4.100-mptcp #1 SMP Wed Feb 24 09:39:10 UTC 2021 x86_64 GNU/Linux
OpenMPTCProuter VPS provider: MilkyWan (they are great !!
Raspberry Pi 4 Model B Rev 1.2

Thanks for your help

@Ysurac can you try to add in your faq how to have omr version (easy, it's in the interface) and the vps version ? I can't remember the command and try to find it but don't find anything

Ysurac, 2022-04-03:
Follow: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding
If still not working, follow debug section of this page.

BillyMyIT, 2022-04-04:
It's ok, thanks for your help.

closed 2022-04-04 by BillyMyIT