Ysurac/openmptcprouter #2314: Hosts file and Hostnames option not working

censorshipm, 2022-04-04:
Hello,
I'm trying to edit the hosts file (/etc/hosts) but it doesn't work.
I'm also trying to use the Hostnames option, but it doesn't seem to work either.
I of course restarted the router.

My hosts file looks like this
```
127.0.0.1 localhost

::1     localhost ip6-localhost ip6-loopback
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters

192.168.0.111 nas.orgg
```

Is there something i'm missing?

## Specifications

  - OpenMPTCProuter version: v0.59beta3-5.4
  - I also tried with this version : v0.58.5-r0
  - OpenMPTCProuter VPS version: 0.1027-test 5.4.132-mptcp
  - OpenMPTCProuter VPS provider: Milkywan
  - OpenMPTCProuter platform: RP4B 64 bits

Ysurac, 2022-04-06:
You can use "Addresses" in Network->DHCP&DNS or Network->Hostnames

censorshipm, 2022-04-06:
Damned, I didn't think about that!
It's working. Many thanks!

closed 2022-04-06 by censorshipm