Ysurac/openmptcprouter #2315: Recommended hardware for adding 4G modem to x86 router

aderk, 2022-04-07:
Are there recommended USB 4G/LTE modems for use with OpenMPTCPRouter on x86? Does OpenMPTCPRouter support MBIM and/or QMI mode for USB modems? Any recommendations for configuration once plugged in?

Options considered:
1) USB modem <-> ROOter <--ethernet--> OpenMPTCPRouter
* Pros: Simple, robust
* Cons: ugg, another appliance to run ROOter =(
2) USB modem <-> OpenMPTCPRouter
* Pros: simple hardware
* Cons: Setting up the modem

This [modem from ThinkPenguin](https://www.thinkpenguin.com/gnu-linux/usb-4g-lte-advanced-modem-gnulinux-tpe-usb4glte) looks like it has good compatibility:
```
Supported Kernels
Under QMI Mode LT4120: Linux 4.16+ & Linux LTS kernels 4.14.44+, 4.4.134+, & 4.9.103+
Under QMI Mode EM7455: Linux 4.7+ & Linux LTS kernels 4.4.16+, 3.16.42+
Under MBIM Mode LT4120: Linux 4.8+
Under MBIM Mode EM7455: Linux 4.2+
```

Thanks, from a proud supporter via Patreon!

Ysurac, 2022-04-07:
MBIM and QMI are supported by OpenMPTCProuter. Most USB modems should work (if not driver can often be installed). In most case it's better to use ModemManager option in OpenMPTCProuter (that support MBIM and QMI)
There is also some modems that give directly an ethernet port like most Huawei 4/5G devices and that even easier to use.

iw5cin, 2022-05-02:
i use an old huawei E8372.
With the network overview
http://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/status
I can also see the level of the LTE signal
congratulations again for the openMPTCProuter project

aderk, 2022-08-01:
Thanks for the information!

closed 2022-08-01 by aderk