Ysurac/openmptcprouter #2317: Can't contact server admin script (no result)

timbrookers, 2022-04-08:
## Expected Behavior

I had setup the openmptcprouter successfully.

## Current Behavior

But the server status said "Can't contact server admin script (no result)".
All others are green.

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: ionos.com
  - OpenMPTCProuter platform: 

[image]

Ysurac, 2022-04-08:
I see that you have a high latency, this may be the reason why you get this message (it timeout for status page).
Else everything is working ? Do you have "openmptcprouter-vps" errors in Status->System log ?

timbrookers, 2022-04-08:
Thank you for reply,
yes, it is working
the error is: 
daemon.err /usr/bin/ss-redir[4507]: remote recv: Connection reset by peer

Ysurac, 2022-04-08:
This error is not related.
In system->OpenMPTCProuter, "Advanced settings" tab you can find "VPS checks timeout", you can try to increase it, this should fix status page display.

timbrookers, 2022-04-08:
ok, i had change it to 20 from 1
now the status is saying localhost

timbrookers, 2022-04-08:
[image]

Ysurac, 2022-04-08:
Ok, it's the hostname of your VPS. Stange hostname, but why not ;)

timbrookers, 2022-04-08:
my download speed is about 102Mbps, but upload is 2.1Mbps
the both isp have download 80Mbps and upload 16Mbps
why it is so?

Ysurac, 2022-04-08:
You can try to change TCP Congestion Control in Network->MPTCP
It's quite difficult to aggregate correctly connections with high latencies.

timbrookers, 2022-04-08:
now i get this error: 
daemon.err /usr/bin/ss-local[4975]: getpeername: Socket not connected

timbrookers, 2022-04-09:
apologies  for disturbing, but
i have new error, 
Can't get Public IP address from Shadowsocks

timbrookers, 2022-04-09:
after restarting the server no issue

thank you

closed 2022-07-13 by github-actions[bot]