Ysurac/openmptcprouter #2321: NanoPI R4S Help

mobizent, 2022-04-13:
I have a NanoPI R4S 4Gb version & i downloaded both the ext4 & sqashfs versions & tried to boot. unfortunately i'm not able to get to the web ui as i dont get an dhcp address. i even tried setting a static ip in the subnet but no luck. any idea what i'm doing wrong?

Ysurac, 2022-04-13:
Please use the template.
I need more info: version used, how you put image on sd card, what static ip you tried,...

mobizent, 2022-04-13:
thanks

Image used: openmptcprouter-v0.58.5-r0+16336-b36068d35d-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz

Used balena etcher to write to microSD

static IP i used was 192.168.100.10 with gateway as 192.168.100.1 & subnet mask 255.255.255.0

mstrolo, 2022-04-25:
I installed I and it's working.

i need help regarding the USB its not getting recognized - I connected USB modem

do you have any idea?

TalalMash, 2022-05-15:
Try a different SD card.

closed 2022-08-18 by github-actions[bot]