Ysurac/openmptcprouter #2325: Tried to change to new VPS and now network settings seem borked?

Fffrank, 2022-04-15:
## Expected Behavior
Erase old VPS and install new VPS without any connectivity issues.

## Current Behavior
Deleted old VPS in Wizard and added new one -- can't find any evidence it is pulling keys or connecting at all (nothing in system log.)  Status Page says Network Overview: No Data and never comes up with the visualization.

If I delete the VPS entirely from the system then it does display the visual on the Status page but obviously it has nothing to connect to.

## Specifications

  - OpenMPTCProuter version: 0.59b3
  - OpenMPTCProuter VPS version: .1027 test
  - OpenMPTCProuter VPS provider: Have tried original VPS (hostworld) as well as Oracle Cloud and BuyVM -- all w/ the same results.
  - OpenMPTCProuter platform: rpi4

Ysurac, 2022-04-15:
In System->OpenMPTCProuter, in the wizard check "show advanced settings" checkbox and check "Force retrieve settings" then save the wizard.

Fffrank, 2022-04-15:
No difference and nothing in logs that I see.

Ysurac, 2022-04-15:
You can check in same part of the wizard if you get same keys as you get in ```/root/openmptcprouter_config.txt``` on the VPS.
Check also with ```uname -a``` on the VPS that you have a mptcp kernel.

Fffrank, 2022-04-15:
I wonder if that's the issue -- here is what I get from `uname -a`

`Linux localhost 4.19.0-18-amd64 #1 SMP Debian 4.19.208-1 (2021-09-29) x86_64 GNU/Linux`

Ysurac, 2022-04-15:
This is not mptcp kernel. Did you reboot VPS after running the script ?

Fffrank, 2022-04-15:
I re-ran the script as well as rebooted twice.  No change.  Any other ideas?

Ysurac, 2022-04-15:
You don't have any errors when you run the script ?
What is the Debian release ?

Fffrank, 2022-04-15:
Debian 10

Here is the only error I see: 
```
/etc/sysctl.d/90-shadowsocks.conf  100%[================================================================>]   1.77K  --.-KB/s    in 0s

2022-04-15 12:03:05 (39.2 MB/s) - '/etc/sysctl.d/90-shadowsocks.conf' saved [1811/1811]

--2022-04-15 12:03:05--  https://www.openmptcprouter.com/server-test/shadowsocks-libev-manager@.service.in
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 68.70.205.4, 68.70.205.1, 68.70.205.3, ...
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|68.70.205.4|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 386 [application/octet-stream]
Saving to: '/lib/systemd/system/shadowsocks-libev-manager@.service'

/lib/systemd/system/shadowsocks-li 100%[================================================================>]     386  --.-KB/s    in 0s

2022-04-15 12:03:05 (8.80 MB/s) - '/lib/systemd/system/shadowsocks-libev-manager@.service' saved [386/386]

Failed to get unit file state for shadowsocks-libev.service: No such file or directory
--2022-04-15 12:03:05--  https://www.openmptcprouter.com/server-test/omr-update.service.in
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 68.70.205.4, 68.70.205.1, 68.70.205.3, ...
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|68.70.205.4|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 337 [application/octet-stream]
Saving to: '/lib/systemd/system/omr-update.service'
```

Ysurac, 2022-04-15:
no real error here.
What do you have in your ```/boot/grub/grub.cfg``` ? what is the result of ```ls /boot``` ?

Fffrank, 2022-04-15:
```
root@localhost:~# cat /boot/grub/grub.cfg
#
# DO NOT EDIT THIS FILE
#
# It is automatically generated by grub-mkconfig using templates
# from /etc/grub.d and settings from /etc/default/grub
#

### BEGIN /etc/grub.d/00_header ###
if [ -s $prefix/grubenv ]; then
  set have_grubenv=true
  load_env
fi
if [ "${next_entry}" ] ; then
   set default="${next_entry}"
   set next_entry=
   save_env next_entry
   set boot_once=true
else
   set default="0"
fi

if [ x"${feature_menuentry_id}" = xy ]; then
  menuentry_id_option="--id"
else
  menuentry_id_option=""
fi

export menuentry_id_option

if [ "${prev_saved_entry}" ]; then
  set saved_entry="${prev_saved_entry}"
  save_env saved_entry
  set prev_saved_entry=
  save_env prev_saved_entry
  set boot_once=true
fi

function savedefault {
  if [ -z "${boot_once}" ]; then
    saved_entry="${chosen}"
    save_env saved_entry
  fi
}
function load_video {
  if [ x$feature_all_video_module = xy ]; then
    insmod all_video
  else
    insmod efi_gop
    insmod efi_uga
    insmod ieee1275_fb
    insmod vbe
    insmod vga
    insmod video_bochs
    insmod video_cirrus
  fi
}

if [ x$feature_default_font_path = xy ] ; then
   font=unicode
else
insmod part_msdos
insmod ext2
if [ x$feature_platform_search_hint = xy ]; then
  search --no-floppy --fs-uuid --set=root  dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
else
  search --no-floppy --fs-uuid --set=root dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
fi
    font="/usr/share/grub/unicode.pf2"
[... 126 lines cut ...]
### END /etc/grub.d/30_os-prober ###

### BEGIN /etc/grub.d/30_uefi-firmware ###
### END /etc/grub.d/30_uefi-firmware ###

### BEGIN /etc/grub.d/40_custom ###
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.
### END /etc/grub.d/40_custom ###

### BEGIN /etc/grub.d/41_custom ###
if [ -f  ${config_directory}/custom.cfg ]; then
  source ${config_directory}/custom.cfg
elif [ -z "${config_directory}" -a -f  $prefix/custom.cfg ]; then
  source $prefix/custom.cfg;
fi
### END /etc/grub.d/41_custom ###
root@localhost:~# ls /boot
config-4.19.0-18-amd64  grub                        initrd.img-5.4.132-mptcp    System.map-5.4.132-mptcp  vmlinuz-5.4.132-mptcp
config-4.19.234.mptcp   initrd.img-4.19.0-18-amd64  System.map-4.19.0-18-amd64  vmlinuz-4.19.0-18-amd64
config-5.4.100-mptcp    initrd.img-4.19.234.mptcp   System.map-4.19.234.mptcp   vmlinuz-4.19.234.mptcp
config-5.4.132-mptcp    initrd.img-5.4.100-mptcp    System.map-5.4.100-mptcp    vmlinuz-5.4.100-mptcp

```

Ysurac, 2022-04-15:
grub.cfg wasn't modified by the install script. I will check why...

Fffrank, 2022-04-15:
Thanks. Let me know if the entire install log would be helpful?

Ysurac, 2022-04-16:
What is the result of ```apt-get -y install linux-image-5.4.132-mptcp=1.19+4f508aa``` ?

Fffrank, 2022-04-16:
```
root@localhost:~# apt-get -y install linux-image-5.4.132-mptcp=1.19+4f508aa
Reading package lists... Done
Building dependency tree
Reading state information... Done
linux-image-5.4.132-mptcp is already the newest version (1.19+4f508aa).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
root@localhost:~#
```

Ysurac, 2022-04-16:
i'm not able to reproduce the issue on VPS providers I use, what is the provider you use ? (if you can reproduce on one I can pay with paypal it would be great)

Fffrank, 2022-04-17:
The provider I'm using is buyvm.net -- I am going to start from scratch and see if any changes.

I've set up a new account on my oracle cloud and reflashed the image to factory reset.  It's now working.  Thanks!

closed 2022-07-21 by github-actions[bot]